riscv-software-src/riscv-isa-sim #746: when mstatus.MPRV=1,mstatus.MPP=3, in machine mode, data and code on the same page will cause a problem

chsgcxy, 2021-07-14:
# when mstatus.MPRV=1,mstatus.MPP=3, in machine mode, data and code on the same page will cause a problem

In chapter 3.1.6.3 with RISC-V privileged Architecture document version 1.12-draft which created May 24 2021, I found the description: When MPRV=1, load and store memory addresses are translated and protected, and endianness is applied, as though the current privilege mode were set to MPP. Instruction address-translation and protection are unaffected by the setting of MPRV.

so, I creat a case to ***test whether MPRV is work and not effact instructon address-translation***. 

## my case

in my case, I put code and data in one page, that cause a problem when run spike.
***so there is a problem in mmu ? Or software should never put code and data in the same page when set mstatus.MPRV=1***

I set mstatus.MPRV=1 and mstatus.MPP=3 in this case, the running mode is machine, the code and data are on the same page which can be loaded/stored/executed. I found that a store(or load) inst can use page table to translate, but after store(or load) inst completed, processor fetches inst using the same VA to PA translation that store(or load) created before.  Processor fetches inst should unaffect by the MPRV, it should use VA=PA according to privilege document.

***next is my test code in riscv-tests repo, and use env/p, link start at 0x80000000***

```asm
#include "riscv_test.h"
#include "test_macros.h"

#define PAGE_TABLE1 0x80004000
#define PAGE_TABLE2 0x80005000
#define PAGE2 0x80002000

RVTEST_RV64M
RVTEST_CODE_BEGIN

page0:
  # Turn on MMU
  li a0, (SATP_MODE & ~(SATP_MODE<<1)) * SATP_MODE_SV39
  la a1, page_table0
  srl a1, a1, RISCV_PGSHIFT
  or a1, a1, a0
  csrw sptbr, a1
  sfence.vma

  # Set up MPRV with MPP=S, so loads and stores use S-mode
  li a1, ((MSTATUS_MPP & ~(MSTATUS_MPP<<1)) * PRV_U) | MSTATUS_MPRV
  csrs mstatus, a1

  j page1

  TEST_PASSFAIL

RVTEST_CODE_END

.align 12
page1:
  # test code, run some random inst
  # then run a load inst to trip page table translate
  add t0, x0, a0
  sub t0, x0, a1
  li t0, 0x80001000
  lw t1, (t0)
  li TESTNUM, 1
  RVTEST_PASS

.align 12
page2:
  .word 0x11111111
  .word 0x22222222
  mul t1, x0, t0
  mul t1, x0, t0
  mul t1, x0, t0
  mul t1, x0, t0
  mul t1, x0, t0
  li TESTNUM, 2
  RVTEST_FAIL

# page table can convert VA:0x80001000 to page2(PA:0x80002000)
.align 12
page_table0:
 .dword 0
 .dword 0
 # vpn[2] = 2
 .dword ((PAGE_TABLE1 >> RISCV_PGSHIFT) << PTE_PPN_SHIFT) | PTE_V

.align 12
page_table1:
 # vpn[1] = 0
 .dword ((PAGE_TABLE2 >> RISCV_PGSHIFT) << PTE_PPN_SHIFT) | PTE_V

.align 12
page_table2:
 .dword 0
 # vpn[0] = 1
 .dword ((PAGE2 >> RISCV_PGSHIFT) << PTE_PPN_SHIFT) | PTE_V | PTE_R | PTE_W | PTE_X | PTE_U | PTE_A

  .data
RVTEST_DATA_BEGIN

  TEST_DATA

dummy: .dword 0

RVTEST_DATA_END
```

## case log

***next is spike log, I add a commit log to mmu walk, we can see that it execute a mul inst in page2, but it should execute "li TESTNUM, 1" in page 1***

```text
core   0: >>>>  page1
core   0: 0x0000000080001000 (0x00a002b3) add     t0, zero, a0
core   0: 3 0x0000000080001000 (0x00a002b3) x5 0x8000000000000000
core   0: 0x0000000080001004 (0x40b002b3) sub     t0, zero, a1
core   0: 3 0x0000000080001004 (0x40b002b3) x5 0xfffffffffffe0000
core   0: 0x0000000080001008 (0x000802b7) lui     t0, 0x80
core   0: 3 0x0000000080001008 (0x000802b7) x5 0x0000000000080000
core   0: 0x000000008000100c (0x0012829b) addiw   t0, t0, 1
core   0: 3 0x000000008000100c (0x0012829b) x5 0x0000000000080001
core   0: 0x0000000080001010 (0x00c29293) slli    t0, t0, 12
core   0: 3 0x0000000080001010 (0x00c29293) x5 0x0000000080001000
core   0: 0x0000000080001014 (0x0002a303) lw      t1, 0(t0)
core   0: mmu stage1 walk vaddr 0x80001000
core   0: mmu stage1 walk level 2 index 2 pte addr 0x80003010 pte 0x20001001(next base 0x80004000)
core   0: mmu stage1 walk level 1 index 0 pte addr 0x80004000 pte 0x20001401(next base 0x80005000)
core   0: mmu stage1 walk level 0 index 1 pte addr 0x80005008 pte 0x2000085f(next base 0x80002000)
core   0: mmu vaddr 0x80001000 get paddr 0x80002000
core   0: 3 0x0000000080001014 (0x0002a303) x6 0x0000000011111111 mem 0x0000000080001000
core   0: 0x0000000080001018 (0x02500333) mul     t1, zero, t0
core   0: 3 0x0000000080001018 (0x02500333) x6 0x0000000000000000
core   0: 0x000000008000101c (0x00004189) c.li    gp, 2
core   0: 3 0x000000008000101c (0x4189) x3 0x0000000000000002
```

## need to fix ?

in refill_tlb, although we use 3 tag(tlb_load_tag, tlb_store_tag, tlb_insn_tag) to make a distinction between load, store and execute, but we only use one tlb_data, when we execute a load or store in this page, a new tlb_entry_t is fill, so when load_insn in a slow path, spike will translate_insn_addr use a tlb_data that load or store inst filled.

so, ***do we need to fix this problem*** ?

If necessary, I can make a fix to this. I think use a itlb[TLB_ENTRIES] and dtlb[TLB_ENTRIES]  to instead tlb_data[TLB_ENTRIES] will work.

chsgcxy, 2021-07-16:
I would be very grateful if anyone could reply @arunthomas @gkelly @scottj97 @sorear

aswaterman, 2021-07-16:
@-mentioning a bunch of people is kind of obnoxious.

chsgcxy, 2021-07-16:
soooooooooo sorry

chsgcxy, 2021-07-16:
> @-mentioning a bunch of people is kind of obnoxious.

soooooooooo sorry

aswaterman, 2021-07-16:
I am currently studying this and agree it seems like a bug in Spike.  I think it can be solved in a less brute-force way than what you've proposed, but either way, I will handle it.  Thanks for bringing it to our attention.

chsgcxy, 2021-07-16:
thanks aswaterman !

aswaterman, 2021-07-16:
It turns out an easier solution was to just avoid refilling the TLB while MPRV is set.

Thanks again for pointing this out and providing a detailed bug report.  Being able to reuse your test saved me a bunch of time.

commit 4506ac36 (2021-07-16): Fix MPRV-related bug

The wrong instruction might've been fetched when the PC was on the same
page as a load or store used within the MPRV sequence.

Fix by not using TLB within MPRV sequences.

Resolves #746
closed 2021-07-16 by aswaterman