riscv-software-src/riscv-isa-sim #747: A question about reporting exception type when access user counter.

tianhu-zeng, 2021-07-16:
[image]
[image]
According to Spec description:
1. when mcounteren[0]=1 & hcounteren[0]=0&scounteren[0]=1, read cycle csr should report virtual instruction exception on VU mode. Spike report the same exception type, it's same with Spec.
2. when mcounteren[0]=1 & hcounteren[0]=0, read cycle csr should also report virtual instruction exception on VU mode. But spike report illegal instruction exception. It seems that the Spike mismatch with the Spec.

@avpatel Please have a look, thanks.

tianhu-zeng, 2021-07-21:
@avpatel If you have time, please have a look, thanks.

avpatel, 2021-07-21:
Yes, I have noted this issue. I try to fix this once I am done with AIA software efforts (may be afi another month or so).

Regards,
Anup

tianhu-zeng, 2021-07-21:
Thank you for your replay!

Tianhu

tianhu-zeng, 2021-08-31:
@avpatel Do you have any plans to fix this issue?

Regards,
Tianhu

avpatel, 2021-08-31:
Sorry for the delay, just finished AIA proof-of-concept last week.

Here's the PR to fix this issue:
https://github.com/riscv/riscv-isa-sim/pull/788

I don't have the exact test which you have so can you please verify this PR.

Regards,
Anup

scottj97, 2021-09-05:
@tianhu-zeng can we close this issue now?

tianhu-zeng, 2021-09-06:
Yes, we can.

closed 2021-09-06 by tianhu-zeng