riscv-software-src/riscv-isa-sim #756: rvp: spike does not check legality of Zpsfoperand insns on rv32

marcfedorow, 2021-07-26:
https://github.com/riscv/riscv-p-spec/blob/master/P-ext-proposal.adoc#detailed-instruction-descriptions-both-rv32-rv64
Spec says that rv32-Zpsfoperand registers are even, unaligned (odd) are reserved.
I suppose spike should throw an illegal instruction on rv32 if e.g. `SMAL x1, x3, x5` is encoded.

aswaterman, 2021-07-26:
Agreed.  It actually looks like this will result in out-of-bounds array access when one of the register specifiers is x31.  Furthermore, it appears that accesses to `x0` are not behaving as specified: the entire write should be discarded, and the entire read should return zero.

aswaterman, 2021-07-26:
Fixed by 95a0a3a9e3dde1bdb18fe972013cca4cd9b78036

closed 2021-07-26 by aswaterman