riscv-software-src/riscv-isa-sim #770: Does spike support adding custom instructions coded in assembly or C?

nikita890-art, 2021-08-15:
I am trying to develop a custom instruction linke for example mod instruction and add it to the spike simulator and the gnu toolchain. But I am not sure if spike supports an instruction coded in C or assembly.
I personally prefer C but I don't know if spike will support it. 
Does it?

aswaterman, 2021-08-16:
If you intend on modifying Spike, you should allocate several hours to read the source code.  In doing so you'll see that Spike models instructions using C++.  (Inline assembly is possible but inadvisable.)

nikita890-art, 2021-08-17:
Why is inline assembly inadvisable? and what is the difference between inline assembly and normal assembly language?
I am planning to code the instruction based on this link,
https://en.cppreference.com/w/c/language/asm

scottj97, 2021-09-05:
Inline assembly is not portable, and for modeling an instruction's semantics in Spike it's unnecessary. Do you really want to describe your `mod` instruction in terms of x86 assembly instructions?

See [how add works](https://github.com/riscv/riscv-isa-sim/blob/master/riscv/insns/add.h)? You can do `mod` using the same approach. Change the `+` to `%` and the compiler will take care of everything.

"Inline" assembly refers to assembly code embedded inside another language (such as C) which the C compiler will recognize, versus an entire file of only assembly language code which an assembler can read.

It might be useful to use inline assembly to write RISC-V code that uses your new `mod` instruction, unless/until you teach GCC when & how to use that new instruction.

NazerkeT, 2023-01-17:
Hi there!

@aswaterman @scottj97 I have the same question as the OP here :)

I was following this [tutorial](https://quasilyte.dev/blog/post/riscv32-custom-instruction-and-its-simulation/) to test new instructions via spike, but noticed that the output is not as expected when I used `.word` based inline assembly. Can spike still work with .word inline assemblies? Or should go and change the binutils to be able to insert the assembly code?

Ideally, instead of rebuilding the toolchain every time for new instruction, I would like to add the new instruction to the C code without any gcc/gas changes, do you think it is possible? Could you please give some hints on it?

Thanks,
Best wishes,

aswaterman, 2023-01-17:
Spike doesn't ever see your assembly code--it only sees the machine code that has been assembled then linked into an ELF executable.  It doesn't matter how you write the code, so long as it forms a correct program.  Modifying GAS thus isn't necessary if you don't mind hand-assembling the code.  Most likely, you have a bug either in your assembly code (or in your implementation of a custom instruction).

NazerkeT, 2023-01-17:
@aswaterman Thanks for the info!

If you have some time, do you think there is smth wrong with this code and assembly?

```
#include <stdio.h> 
// Needed to verify results.
int mac_c(int a, int b, int c) {
    a += b * c; // Semantically, it is "mac"
    return a;
}
// Should not be inlined, because we expect arguments 
// in particular registers.
__attribute__((noinline))
int mac_asm(int a, int b, int c) {
    asm __volatile__ (".word 0x02C5856B\n");
    return a;
}
int main(int argc, char** argv) {
    int a = 2, b = 3, c = 4;
    printf("%d =?= %d\n", mac_c(a, b, c), mac_asm(a, b, c));
}
```

```
00000000000101a2 <mac_c>:
   101a2:       02c585bb                mulw    a1,a1,a2
   101a6:       9d2d                    addw    a0,a0,a1
   101a8:       8082                    ret

00000000000101aa <mac_asm>:
   101aa:       02c5856b                .word   0x02c5856b
   101ae:       8082                    ret

00000000000101b0 <main>:
   101b0:       1141                    addi    sp,sp,-16
   101b2:       e406                    sd      ra,8(sp)
   101b4:       4611                    li      a2,4  
   101b6:       458d                    li      a1,3  
   101b8:       4509                    li      a0,2  
   101ba:       ff1ff0ef                jal     ra,101aa <mac_asm>
   101be:       862a                    mv      a2,a0 
   101c0:       45b9                    li      a1,14 
   101c2:       6571                    lui     a0,0x1c
   101c4:       79050513                addi    a0,a0,1936 # 1c790 <__clzdi2+0x44>
   101c8:       142000ef                jal     ra,1030a <printf>
   101cc:       4501                    li      a0,0  
   101ce:       60a2                    ld      ra,8(sp)
   101d0:       0141                    addi    sp,sp,16
   101d2:       8082                    ret
```

Thanks,

aswaterman, 2023-01-17:
You either need to write the function in pure assembly, or you need to use constraints on the asm block so that the compiler understands the dataflow. The general concept isn’t RISC-V-specific, so you should be able to find tutorials elsewhere online.

NazerkeT, 2023-01-18:
Got it, thanks for a quick response! 

I will put a link to my solution if i find some here :)

Best wishes,