riscv-software-src/riscv-isa-sim #771: cannot run vector isa extention in linux-5.13

vigoss30611, 2021-08-18:
I ran V ISA on spike and pk successfully.

riscv64-unknown-linux-gnu-gcc -c test.c -o helloWorld
spike --isa=RV64IMAFDCV pk helloWorld

test.c:
```
#include <stdio.h>

int main()
{
  printf("hello world\n");
  asm volatile("vsetvli a4, a0, e32, m8, ta, ma");
  printf("hello world\n");
  return 0;
}
```

and i ran linux-5.13 on spike successfully,and when I get correct cpuinfo:
```
cat /proc/cpuinfo
processor	: 0
hart		: 0
isa		: rv64imafdcv
mmu		: sv48
```

spike --isa=RV64IMAFDCV bbl

in linux(busybox)
./helloWorld

error
```
hello world
[   33.316060] helloWorld[47]: unhandled signal 4 code 0x1 at 0x00000000000105fa in helloWorld[10000+5a000]
[   33.316875] CPU: 0 PID: 47 Comm: helloWorld Not tainted 5.13.11 #13
[   33.317375] Hardware name: ucbbar,spike-bare (DT)
[   33.317695] epc : 00000000000105fa ra : 00000000000105fa sp : 0000003fffbcfca0
[   33.318405]  gp : 000000000006d1a8 tp : 000000000006f710 t0 : 000000000006b648
[   33.319065]  t1 : 2f2f2f2f2f2f2f2f t2 : 0000000000001000 s0 : 0000003fffbcfcb0
[   33.319725]  s1 : 0000000000010b4a a0 : 0000000000000011 a1 : 00000000000707c0
[   33.320385]  a2 : 0000000000000011 a3 : 00000000000707c0 a4 : 0000000000000001
[   33.321045]  a5 : 0000000000000001 a6 : 0000000000000000 a7 : 0000000000000040
[   33.321705]  s2 : 0000000000000000 s3 : 0000000000155fe8 s4 : 00000000001b0bc0
[   33.322410]  s5 : 0000000000000008 s6 : 00000000001af798 s7 : 0000000000000014
[   33.323070]  s8 : 00000000001b0b10 s9 : 00000000001af798 s10: 00000000001b0bb0
[   33.323730]  s11: 0000000000000001 t3 : ffffffffffffffff t4 : 0000000000000040
[   33.324390]  t5 : 0000000000000000 t6 : 0000000000000004
[   33.324810] status: 0000000200004020 badaddr: 0000000002008157 cause: 0000000000000002
Illegal instruction
```

chihminchao, 2021-08-18:
status: 0000000200004020
It seems the mstatus.vs (mstatus[10:9]) bit is not set and the field needs to be non-zero value to enable vector core.

aswaterman, 2021-08-18:
Yeah, not a Spike bug.  Sounds like you are running a Linux kernel that is not vector-aware.

closed 2021-08-18 by aswaterman

vigoss30611, 2021-08-18:
> ```
>  status: 0000000200004020
> ```
> 
> It seems the mstatus.vs (mstatus[10:9]) bit is not set and the field needs to be non-zero value to enable vector core.

It seems that. But I find In riscv-pk, mstatus[10:9] have set 0x3 in function (mstatus_init),and I can run v isa extension(in 
linux-5.13, like that:
asm volatile("vsetvli a4, a0, e32, m8, ta, ma");
even if I add this before run_init_process() (in init/main.c).
But I cannot run it in user space. Maybe somewhere in Linux kernel change the value of mstatus, but I cannot find out it...
Can someone help me!