riscv-software-src/riscv-isa-sim #787: Which instruction set combination can support a simple Linux operating system

LangQingjie, 2021-08-31:
Hello, I am preparing to develop a processor core that can support Linux operating system in order to complete my course homework. But now I'm very confused because I don't know which instruction set is the least needed to support a Linux operating system.

aswaterman, 2021-08-31:
At bare minimum, the Linux kernel requires the RV32IMA instruction set, but in practice, distros require RV64IMAFDC.  In both cases, a lot of privileged architecture support is also needed.

LangQingjie, 2021-08-31:
sorry,  I have closed the issue on riscv/riscv-tools.

LangQingjie, 2021-08-31:
Hello, does privileged architecture support refer to privilege level and debug mode?

aswaterman, 2021-08-31:
Debug is not necessary to run Linux.