riscv-software-src/riscv-isa-sim #799: Spike segmentation fault when executing OpenSBI

jolivain, 2021-09-09:
Spike is crashing with segmentation fault when executing OpenSBI.

## How to reproduce:

### Step 1: compile OpenSBI

Assuming a Risc-V toolchain is in the PATH:
```
git clone https://github.com/riscv-software-src/opensbi
cd opensbi
make -j$(nproc) CROSS_COMPILE=riscv64-unknown-elf- PLATFORM=generic
```
Tested at commit bd355213bfbb209c047e8cc0df56936f6705477f

### Step 2: compile riscv-isa-sim

```
git clone https://github.com/riscv-software-src/riscv-isa-sim
cd riscv-isa-sim
mkdir build
cd build
../configure
make -j$(nproc)
```
Tested at commit 83502ad4afd3e3387aed4c45b290d00fd2fbc2ee

### Step 3: run

```
./spike -l /path/to/opensbi/build/platform/generic/firmware/fw_payload.elf
...
core   0: 0x0000000080007ce6 (0x00068733) add     a4, a3, zero
core   0: 0x0000000080007cea (0x30561673) csrrw   a2, mtvec, a2
core   0: 0x0000000080007cee (0x3bf59073) csrw    pmpaddr15, a1
```
In GDB:
```
Program received signal SIGSEGV, Segmentation fault.
0x0000000000772fbb in pmpaddr_csr_t::next_locked_and_tor (this=0x8f88c0) at ../riscv/csrs.cc:120
120	  bool next_locked = state->pmpaddr[pmpidx+1]->cfg & PMP_L;

Backtrace:
#0  0x0000000000772fbb in pmpaddr_csr_t::next_locked_and_tor (this=0xa8ea70) at ../riscv/csrs.cc:120
#1  0x0000000000773018 in pmpaddr_csr_t::unlogged_write (this=0xa8ea70, val=0) at ../riscv/csrs.cc:109
#2  0x0000000000772e9a in csr_t::write (this=0xa8ea70, val=<optimized out>) at ../riscv/csrs.cc:44
#3  0x000000000057d29d in processor_t::set_csr (this=this@entry=0x9db4e0, which=which@entry=959, val=<optimized out>) at ../riscv/processor.cc:916
#4  0x0000000000772291 in rv64_csrrw (p=0x9db4e0, insn=..., pc=2147517708) at ../riscv/insns/csrrw.h:3
#5  0x000000000077fb50 in execute_insn (p=0x9db4e0, pc=2147517708, fetch=...) at ../riscv/execute.cc:180
#6  0x000000000077fd33 in processor_t::step (this=0x9db4e0, n=4530) at ../riscv/execute.cc:289
#7  0x00000000005a026c in sim_t::step (this=this@entry=0x7fffffffc700, n=n@entry=5000) at /usr/include/c++/11/bits/stl_vector.h:1043
#8  0x000000000055ff4c in sim_t::main (this=0x7fffffffc700) at ../riscv/sim.cc:187
#9  0x00007ffff798a990 in ?? () from /lib64/libc.so.6
#10 0x0000000000a8f150 in ?? ()
#11 0x0000000000000000 in ?? ()
```
Spike crash here:
https://github.com/riscv-software-src/riscv-isa-sim/blob/83502ad4afd3e3387aed4c45b290d00fd2fbc2ee/riscv/csrs.cc#L120

When opensbi executes here:
https://github.com/riscv-software-src/opensbi/blob/9c9b4ad24bf5891272679a6e994c64463db8ca03/lib/sbi/sbi_hart.c#L449

Expected result: no crash

Thanks!

aswaterman, 2021-09-09:
Thanks for the bug report.  Looks like a recent regression introduced by @scottj97's recent pull request.  In the meantime, if you go back in time (e.g. to this commit probably a04da860635b4e94fc05f23f75fd99578258bc3e) you should have success.

scottj97, 2021-09-09:
#798 fixes

aswaterman, 2021-09-09:
Resolved by merging #798

closed 2021-09-09 by aswaterman