riscv-software-src/riscv-isa-sim #803: A bug about writing scause

wzy971210, 2021-09-15:
Hi!
There's a bug about writing scause. 
Since it is a WLRL csr, only the supported exception codes can be written. 
However when I try to write a random value to scause and read it then, the return value is just the value of write. The following pictures explain the problem.
[image]
[image]

scottj97, 2021-09-15:
The privileged spec says about scause:

> The Exception Code is a WLRL field. It is required to hold the values 0–31 (i.e., bits 4–0 must be implemented), but otherwise it is only guaranteed to hold supported exception codes.

It does not say that only the supported exception codes can be written -- in fact it explicitly requires that some of the unsupported exception codes (e.g. 10 and 11) be writable.

I see no issue here.

aswaterman, 2021-09-15:
Agreed with Scott.  The behavior you described corresponds to WARL, not WLRL.

closed 2021-09-15 by aswaterman

wzy971210, 2021-09-16:
Thanks, I misunderstood before.
So for the WLRL csr, software is needed to ensure the value of write is legal. If an illegal value is written, it still can be read since hardware will not limit the behavior of write.
And for WARL csr, only the legal value can be read whether the the write is legal or not.

aswaterman, 2021-09-16:
For WLRL, hardware _might or might not_ limit the behavior of the write.  So e.g. it would be legal for HW to implement WLRL like Spike does, or it would be legal to implement WLRL using WARL instead.