riscv-software-src/riscv-isa-sim #805: Bad isa option 'rv64gch'

Vasanvs, 2021-09-17:
In the latest Spike commit version, if we try to invoke spike as follows:

spike --isa rv64gch <ELF payload file path>

We get an error message reported back as below before exiting:

error: bad --isa option 'rv64gch'. unsupported extension: h

Has the Hypervisor H extension changed during the last checkin for naming convention?

aswaterman, 2021-09-17:
@marcfedorow your #802 has broken this.  Please accept 'h' as a means to enable the hypervisor extension.

aswaterman, 2021-09-17:
@Vasanvs @neelgala in the mean time, please discard the most recent commit on master.

aswaterman, 2021-09-17:
@marcfedorow I've fixed it myself.  The issue is that you were taking the spec literally, that H is a prefix rather than an extension name.  Unfortunately, the spec is out of date in this regard.  In fact, that chapter needs to be completely rewritten...

commit b4f76106 (2021-09-17): Fix parsing of isa string for H extension

Closes #805
closed 2021-09-17 by aswaterman