riscv-software-src/riscv-isa-sim #809: inconsistencies with xRET instructions

neelgala, 2021-09-25:
The new hypevisor spec draft under public review mentions that when mret is executed mstatus.mpv must be set to 0. However, in [mret.h](https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/insns/mret.h) this is not happening. 

Also in mret.h we have the following which resets mstatus.mprv but the section 5.6.4 in the spec does not indicate reseting mprv:
https://github.com/riscv-software-src/riscv-isa-sim/blob/ce42f1b55a3fa5a3a522439b5370be2256f68862/riscv/insns/mret.h#L6-L7

The above observations also hold for sret instructions. Now while this does seem inconsistent to me, I have been successfull in running a few hypervisor related codes on this. So I am probably missing something obvious here ?
@avpatel @scottj97

neelgala, 2021-09-25:
Also, sret.h is trying to reset mprv in sstatus. It should be reseting mprv in mstatus instead ?

aswaterman, 2021-09-25:
The M-mode chapter says "An MRET or SRET instruction that changes the privilege mode to a mode less privileged than M also sets MPRV=0", so that aspect is correct.  The hypervisor chapter doesn't override this behavior; it just doesn't make mention of it.

The MPV/SPV thing seems like a bug.  I've pushed a fix, which @avpatel and/or @scottj97 should sanity-check.  c8aff0ebfec37ceec04500a2aac9dd0a48610ad8

aswaterman, 2021-09-25:
> Also, sret.h is trying to reset mprv in sstatus. It should be reseting mprv in mstatus instead ?

Yes, that's right, since MPRV is not visible via sstatus.  Fixed via 12131ba08c90045efae33067b79d15f4d2fb63f0

aswaterman, 2021-09-25:
BTW, the SRET/MPRV bug I've just fixed is a new regression introduced in @scottj97's refactoring in c3544f5b11275449e95100855f4d3d466d6b54cb -- it is not something that has been lurking for years.

avpatel, 2021-09-25:
The MSTATUS.MPV and HSTATUS.SPV mainly inform software in given privilege mode whether we came from virtualized mode or not. Apart from this, it also affects behaviour of unpriv access instructions.

Setting MPV and SPV to zero upon xRET instruction does not have much impact on  M-mode firmware and hypervisor but it is certainly required for correctness.

Regards,
Anup

neelgala, 2021-09-25:
@aswaterman I think this

> > Also, sret.h is trying to reset mprv in sstatus. It should be reseting mprv in mstatus instead ?
> 
> Yes, that's right, since MPRV is not visible via sstatus. Fixed via [12131ba](https://github.com/riscv-software-src/riscv-isa-sim/commit/12131ba08c90045efae33067b79d15f4d2fb63f0)

I think this wrong. You should still be changing sie, spie and spp in sstatus (which is vritualized) since it could update vsstatus when virt==1 or sstatus (shadow of mstatus) when virt==0. However, the mprv change should be done in mstatus only.

aswaterman, 2021-09-25:
@neelgala As discussed in the comments on 12131ba08c90045efae33067b79d15f4d2fb63f0, this has been fixed in af4beca4daa9fbb100ba87fad789fabdf9848e63

neelgala, 2021-09-25:
oops.. my bad.. should have scrolled further down.. thanks for the quick response and patch :-). Closing this issue then.

closed 2021-09-25 by neelgala

aswaterman, 2021-09-25:
Cool, thanks for reporting.