riscv-software-src/riscv-isa-sim #811: Is it possible to access physical memory in spike from a user program?

Srijith-r2, 2021-09-26:


aswaterman, 2021-09-26:
It depends on how the privileged software has configured memory protection.  If using `pk`, the answer is "no, not directly": all user accesses go through page-based address translation.  If using different privileged software that has set up the PMP registers permissively, and has not enabled paging, then U-mode can access physical memory directly.

Srijith-r2, 2021-09-27:
> It depends on how the privileged software has configured memory protection. If using `pk`, the answer is "no, not directly": all user accesses go through page-based address translation. If using [...]

Yes... am using pk. Any idea how could i do it with pk?

aswaterman, 2021-09-27:
That would involve rewriting large parts of pk.  You're probably better off not using pk at all, running a bare-metal program itself, as with the programs in https://github.com/riscv-software-src/riscv-tests

closed 2021-12-07 by scottj97