riscv-software-src/riscv-isa-sim #817: New release tag since 1.0.0

r-value, 2021-09-28:
`spike` is now being packaged for Arch Linux, but I found that the version `1.0.0` is too old to be built and packaged now.

Since version `1.0.0` is now 920 commits behind the current `master` branch, could you please make a new release tag, just to update the package?

aswaterman, 2021-09-28:
Good point.  I'd like to wait until @scottj97's massive refactoring is complete and well tested, then we will cut another release.  Hopefully that means in the next month or so.

r-value, 2021-12-14:
@aswaterman Any updates?

aswaterman, 2021-12-14:
@scottj97 any concerns about making a 1.1.0 release with current contents?

scottj97, 2021-12-17:
Doesn't seem very stable lately, but maybe it's never stable. I have no concerns.

aswaterman, 2021-12-17:
True enough.

aswaterman, 2021-12-17:
https://github.com/riscv-software-src/riscv-isa-sim/releases/tag/v1.1.0

closed 2021-12-17 by aswaterman