riscv-software-src/riscv-isa-sim #819: Questions about the vector instruction support in Spike

Grubby-CPU, 2021-09-29:
Hi Guys,

  I have a question about the usage of V0 register. In particular, when V0 is used as the mask register, can it still be used as the destination register.
  I found some text related to this in the riscv-v-spec-1.0-frozen file as shown below.

**The destination vector register group for a masked vector instruction cannot overlap the source mask register (v0), unless
the destination vector register is being written with a mask value (e.g., compares) or the scalar result of a reduction. These
instruction encodings are reserved.**

  Yes, Spike reports an illegal warning when it executes instructions such as "vadd.vv v0, v1, v2, v0.t " and works well when “vmfgt.vv v0, v2, v3, v0.t； vfredosum.vs v0, v0, v5, v0.t ；vmsne.vv v0, v0, v5, v0.t” are executed.
  
  However, I have a question about the instruction named "vmsbf.m v0, v1, v0.t ". As mentioned in the document, the vmsbf.m instruction takes a mask register as input and writes results to a mask register. In this case, I think this should be a legal instruction since V0 is written with a mask value. However, Spike reports an illegal warning for this instruction.

Any comments about this?

Many thanks

aswaterman, 2021-09-29:
Spike is correct.  The vector spec says of `vmsbf.m` (https://github.com/riscv/riscv-v-spec/blob/d3127be724d6777ecc7b91460fc146391ca070db/v-spec.adoc#154-vmsbfm-set-before-first-mask-bit): "The destination register cannot overlap the source register and, if masked, cannot overlap the mask register ('v0')."

Grubby-CPU, 2021-10-09:
I just missed such important information...Thanks, aswaterman!