riscv-software-src/riscv-isa-sim #823: vxsat CSR doesn't work with P-ext

marcfedorow, 2021-10-05:
`vxsat` requires V-ext enabled. Also it sets some vector states.
Thus kadd16 is illegal if -march=rv64imp.
@scottj97

scottj97, 2021-10-05:
I was unaware that `kadd16` and friends share some state (vxsat) with the vector extension.

I can see that my commit 347ff31125bee0fadd8e02bbe7a13c672dcf74a5 inadvertently changed the behavior of `kadd16` to take an exception if misa.V=0. But I'm not sure that Spike was ever fully correct here. Reading through the P spec, it seems that `csrrw vxsat` is supported? I can't see that in Spike -- it was always requiring the V extension for that CSR access. Should that be (V || P)?

Second, if -march=rv64gmpv, what should happen to mstatus.VS when `kadd16` modifies `vxsat`? It seems to me that it should set it to Dirty, yes? I don't believe it was doing so prior to 347ff31125bee0fadd8e02bbe7a13c672dcf74a5.

# Proposed change

1. Create `vxsat` CSR and allow access to it via `csrrw` if either misa.V or misa.P are set. (Today it requires misa.V.)

2. When `vxsat` is modified by any instruction, set `mstatus.VS=Dirty` iff misa.V==1. (Today it sets `mstatus.VS=Dirty` unconditionally. Prior to 347ff31125bee0fadd8e02bbe7a13c672dcf74a5 it was only setting it for Vector instructions, not P instructions.)

Does this sound like the correct behavior? I'm not that familiar with either P or V specs.

aswaterman, 2021-10-05:
I think you left out point # 3:

`vxsat` should be inaccessible (by `csrrw` or by a P or V instruction) when `misa.V`=1 and `mstatus.VS`=Off.  In other words, if V is present, P instructions must respect `VS`.

scottj97, 2021-10-05:
cc @chuanhua not sure how this is intended to work

scottj97, 2021-10-06:
> So, P instructions that access `vxsat` should be gated on `misa.V == 0 || mstatus.VS != Off`, and P instructions that write `vxsat` should set `VS` to Dirty if that gating function is satisfied.

That doesn't sound quite right...the V spec says:

> Implementations may have a writable misa.V field. Analogous to the way in which the floating-point unit is handled, the
mstatus.VS field may exist even if misa.V is clear.

So that would mean that we would write mstatus.VS=Dirty even when misa.V==0? (But only if V is in the ISA string?) Is that okay?

Upon further research, I see that Spike does not have a writable misa.V field. So VS doesn't exist in that case.

I am tempted to simply revert my inadvertent change and let others more familiar with with P&V specs deal with these other issues.

aswaterman, 2021-10-06:
I forgot about that clause.  What I wrote is one valid implementation: as the sentence you quoted says, VS _may_ exist when misa.V is cleared, but it isn't required to exist.  I think that keeping VS around will be a more common implementation strategy, though, so it would be pragmatic to do what you suggested.

I suggest you not revert the change and instead do what you've just suggested: condition P instructions that access `vxsat` on `VS` if V is in the ISA string, and set `VS` to Dirty on P instructions that write `vxsat` if V is in the ISA string.  You can use `misa.V` as a proxy since, as you pointed out, it's not mutable.

scottj97, 2021-10-06:
More research: I cannot see the problem @marcfedorow alluded to. I can successfully run kadd16 even with --isa=rv64imcp (i.e., without V). The commit log shows the write to vxsat, and an extra write to mstatus from it trying (unsuccessfully) to set VS.

The only observable difference I can find from my recent changes is that P-instructions writing vxsat now set mstatus.VS=Dirty when misa.V=1, which is a desirable if unintended change.

Marc, can you explain the expected behavior vs the observed behavior? I am unable to find anything that I broke here.

aswaterman, 2021-10-06:
@scottj97 And do P instructions that access `vxsat` check whether VS=Off then trap when misa.V=1?  That seems like the only missing thing from what you've just described.

kito-cheng, 2021-10-06:
misa.V is 0 when zve* extension is enabled, so I think misa.V can't be the condition checking of vxsat?

marcfedorow, 2021-10-06:
Also I think it is incorrect that P_SET_OV macro writes to vxsat even if it should not.
Macro should be something like
```
if (ov) P.VU.vxsat->write(P.VU.vxsat->read());
```
to be consistent with spec.

PR #826 (2021-10-08): Fix sharing of vxsat between V and P extensions
Fixes #823; see discussion there for context.

Behavior changes:

1. Don't write `vxsat` (or set `mstatus.VS=Dirty` if that field exists) unless the instruction actually saturates
2. If `mstatus.VS` field exists (which today in Spike means `misa.V==1`), P-extension instructions that may write `vxsat` will take an illegal instruction exception if `mstatus.VS==Off`
3. When the `--isa` string contains P but not V, doing `csrrw vxsat` will no longer take an illegal instruction exception

cc @marcfedorow @chuanhua