riscv-software-src/riscv-isa-sim #827: Running eg with spike

sundas56, 2021-10-07:
AOA Hi, while running spike  by 32 bit toolchain it gives this error, I've tried many solutions but nothing works.

terminate called after throwing an instance of 'std::runtime_error'

scottj97, 2021-10-07:
You haven't provided enough information for us to understand what you're trying to do. What command did you run that gave the above error? What else did you try that didn't work?

sundas56, 2021-10-08:
by eg I mean hello word C program, I ran "spike pk hello" and get this error message
terminate called after throwing an instance of 'std::runtime_error'

scottj97, 2021-10-08:
There are many possibilities...how did you compile Spike? How did you compile pk? How did you compile hello? What does hello.c look like? A mixup in any of those might cause this.

Was that the complete error message? Normally a `std::runtime_error` includes a `what()` message, like so:
```
$ ./build/spike --device=a,xxx,c
terminate called after throwing an instance of 'std::runtime_error'
  what():  Error parsing device base address.
Aborted (core dumped)
```