riscv-software-src/riscv-isa-sim #834: Undefined reference error in boost when compiling the latest

sammy17, 2021-10-15:
I am getting the following error when I try to compile latest 

```
/share/pkg.7/gcc/9.3.0/install/bin/g++ -L. -Wl,--export-dynamic -L/usr/lib64  -Wl,-rpath,/projectnb2/risc-v/riscv-isa-sim/build/lib  -o spike-dasm spike-dasm.o  libspike_dasm.a  libriscv.a  libdisasm.a -lpthread -ldl -lboost_regex -lboost_system -lpthread  -lboost_system-mt -lboost_regex-mt
libriscv.a(interactive.o): In function `boost::cpp_regex_traits<char>::transform[abi:cxx11](char const*, char const*) const':
/usr/include/boost/regex/v4/cpp_regex_traits.hpp:919: undefined reference to `boost::re_detail::cpp_regex_traits_implementation<char>::transform[abi:cxx11](char const*, char const*) const'
libriscv.a(interactive.o): In function `boost::cpp_regex_traits<char>::transform_primary[abi:cxx11](char const*, char const*) const':
/usr/include/boost/regex/v4/cpp_regex_traits.hpp:923: undefined reference to `boost::re_detail::cpp_regex_traits_implementation<char>::transform_primary[abi:cxx11](char const*, char const*) const'
```

As shown, the error is coming when spike-dasm is being linked. Even if I spike spike-dasm, same error comes when linking spike-main.

Please let me know if there is any way to identify the root cause and resolve the issue.

aswaterman, 2021-10-15:
cc @emelcher — it appears the autoconf checks for Boost aren’t sufficient. (This is part of the reason I was leery of incorporating Boost into Spike to begin with…)

sammy17, 2021-10-15:
It compiled successfully when --without-boost --without-boost-asio --without-boost-regex options were used to configure. Thanks!

aswaterman, 2021-10-29:
I’m absolutely on board with doing better regression testing. I set up a CI flow a couple years ago, but then the CI provider (travis) stopped offering their free service. I haven’t taken the time to migrate the flow to a new CI provider. I’ll look into that soon.

Unfortunately, it’s substantially more difficult to do multi-platform regression testing, as we’d need to do to catch build issues like this one.

It sounds like you’re zeroing in on a fix to the FDT+boost problem, though. That’s good!

timsifive, 2021-10-29:
Since then github has added a feature called "github actions." I'm using it in OpenOCD to verify that builds run, and in a few spec repos to check the PDF build. It works better than travis did IMO.

aswaterman, 2021-10-29:
Thanks, @timsifive; that's probably what we'll do.

emelcher, 2021-11-08:
I noticed that my MacOS Catalina was outdated because issue reports mention Big Sur,
so I updated, but landed directly in Monterey.

#820 has been reproduced and found to be related to a macro definition in riscv/decode.h:177
The name of this macro is the single letter `P`. This is probably not an ideal solution,
considering that there are no namespaces for macros, so conflict with another macro name,
or, as in this case, a template type placeholder name, is not unlikely.
The said macro defines `P` as `(*p)` and then uses it as `P.some_processor_t_property`.
Perhaps using `p->` instead of `P.` would be better and make the definition of that macro
unnecessary?

#829 apparently is that FDT problem which I have already encountered.

#71 apparently is the same as #820.

So, all issues seem to be accounted for and reproduced, so I can start fixing, being able
to test if the fix is effective.

Maybe I should start with the FDT issue, then the P issue and last the config.status issue ?

aswaterman, 2021-11-08:
I'll handle the `P` issue, because I'd like to find a less disruptive fix (if one exists).  The rest of your plan sounds good.

aswaterman, 2021-11-09:
#820 fixed again in #854

aswaterman, 2021-11-10:
Since #856 has also been merged, can we call this issue resolved?

emelcher, 2021-11-11:
As in https://github.com/riscv-software-src/riscv-isa-sim/issues/834#issuecomment-953301742, I suspect that this issue is due to two different versions of boost installed in the user's computer, one installed under `/usr/local` and another under `/usr`, but we seem to have no confirmation from @sammy17 if this is the case or not.
Judging from the error message initially reported for this issue, the FDT fix probably does not resolve this issue.

sequencer, 2022-01-13:
I also encountered this issue today:
```
configure:5866: result: no
configure:5946: checking for library containing dlopen
configure:5977: clang++ -o conftest -g -O2   -Wl,--export-dynamic conftest.cpp -lpthread  >&5
/nix/store/cdm6zywd51mbabxhklsixwcskv4n70s3-binutils-2.35.2/bin/ld: /run/user/1000/conftest-25f9fb.o: in function `main':
/home/user/playground/out/spike/compile/dest/conftest.cpp:41: undefined reference to `dlopen'
```
I was trying to compile under nix environment, however with `-L/lib64` command added by `BOOST_LDFLAGS`, configure directly query the LSB from linux.
adding 
```
 --without-boost --without-boost-asio --without-boost-regex
```
works for me.

Leon924, 2022-04-04:
> 

my linux version is RHEL6.10

I encounter boostlib problem like this. The system already has the version of boostlib 1.41. But I need the version of 1.53 then install it in my homedir. So how should I setup environment path or something else to  make  1.53 version used prior to 1.41?