riscv-software-src/riscv-isa-sim #855: Spike enters infinite loop

pruth111, 2021-11-09:
Hi Team,
I was checking on running TFLM models on riscv. I used a simple hello world model from the TFLM repo and tried to run on Spike.
I used the following command: ** spike -d  proxykernel/riscv32-unknown-elf/bin/pk /home/pvenkatesha/projects/hello_world**
The spike just runs without any info on console. I used -d option and i get the following result:
```
core   0: 0x0000000080000134 (0x000300e7) jalr    t1
: 
core   0: exception trap_instruction_access_fault, epc 0xffffffff800001f8
core   0:           tval 0xffffffff800001f8
: 
core   0: >>>>  trap_vector
core   0: 0x0000000080000004 (0x34011173) csrrw   sp, mscratch, sp
: 
```

aswaterman, 2021-11-09:
Is it possible you’re running a 32-bit proxy kernel on 64-bit spike? You can try adding the --isa=rv32gc flag to spike.

scottj97, 2021-11-09:
I don't know much about pk but this seems to be a very common mistake. Is there a way that Spike could check the ELF it's loading for 32/64 compatibility with the chosen --isa? And similarly could pk check the ELF it's loading?

I can't see a valid use case for trying to run a 32-bit ELF on a 64-bit ISA. But I could be wrong about that.

aswaterman, 2021-11-09:
For systems with writable mstatus.MXL, you could run a 32-bit binary whose first action is to reprogram MXL.  But I think the practical benefit of your suggestion outweighs that hypothetical future concern.

aswaterman, 2021-11-09:
Let me see what plumbing would be necessary to add this check...

pruth111, 2021-11-09:
Thanks @aswaterman. As @scottj97 said, since spike supports both rv64 and 32 , i didn't think of providing isa option. Now it works with this option. I will close the issue.

closed 2021-11-09 by pruth111