riscv-software-src/riscv-isa-sim #860: Getting the clock period of the simulator

vincentabraham, 2021-11-16:
I want to calculate delay between some specific lines of code. So, I have used the following function to retrieve the number of cycles:

```
unsigned long read_cycles(void)
{
  unsigned long cycles;
  asm volatile ("rdcycle %0" : "=r" (cycles));
  return cycles;
}
```

How do I calculate the delay in seconds through the cycles ?

scottj97, 2021-11-16:
Spike does not model time. It only knows instructions, and `rdcycle` actually counts instructions, effectively assuming a CPI of 1. If you make assumptions about CPI and clock frequency then you can calculate time, but those assumptions are beyond the scope of what Spike does.

vincentabraham, 2021-11-17:
> Spike does not model time. It only knows instructions, and `rdcycle` actually counts instructions, effectively assuming a CPI of 1. If you make assumptions about CPI and clock frequency then you can calculate time, but those assumptions are beyond the scope of what Spike does.

So, would the clock frequency be the same as CPU ? Also, if not, would there be any other way to calculate time in SPIKE ?

scottj97, 2021-11-17:
Spike can tell you how many instructions were executed. It does not model a real CPU with a clock frequency. How much time it will take to execute all those instructions will depend on the implementation.

For example, this [SiFive E31](https://www.sifive.com/blog/dhrystone-performance-tuning-on-the-freedom-platform) gets about 0.96 IPC on Dhrystone, so you can expect 100000 instructions to take about 104166 cycles (instructions divided by IPC). But that is only an estimate, since IPC can vary widely with different workloads.

The SiFive E310 datasheet says the E31 is capable of approximately 320 MHz. So 104166 cycles should take about 325 microseconds (cycles divided by MHz).

vincentabraham, 2021-11-18:
Thanks a lot for the information !

closed 2021-11-18 by scottj97

vincentabraham, 2021-11-23:
> Spike can tell you how many instructions were executed. It does not model a real CPU with a clock frequency. How much time it will take to execute all those instructions will depend on the [...]

Is it safe to say that 64-bit RISC-V cores would have the frequency of 1GHz? If not, could you give me some examples of frequencies supported by processors as I can't find any. And would the rdtime instruction work in SPIKE ?

scottj97, 2021-11-26:
RDTIME is an illegal instruction in Spike. It does support an `mtime` memory-mapped register which [does seem to support](https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/clint.cc#L73-L83) some real-time thing where it's measuring the time on the host. I'm not sure how that would be useful.

What are you actually trying to measure? Spike is not designed to be a performance model and it seems like you're trying to do something unreasonable.

vincentabraham, 2021-11-27:
> RDTIME is an illegal instruction in Spike. It does support an `mtime` memory-mapped register which [does seem to [...]

I just want to measure the time taken between some statements of code.

aswaterman, 2021-11-27:
If you're using Newlib on `pk` on spike, then the time.h functions pretend that the target machine has a 1 GHz clock.  If so, you might be observing that `time()` isn't incrementing because you aren't executing enough instructions (you'd need to execute 1 billion of them to see an increment).  Instead, you can use `gettimeofday()`, which has microsecond resolution.  Here is an example:

```
$ cat test.c
#include <sys/time.h>
#include <stdio.h>

float time_in_secs()
{
  struct timeval tv;
  gettimeofday(&tv, NULL);
  return tv.tv_sec + 1e-6f * tv.tv_usec;
}

int main()
{
  float t0 = time_in_secs();

  // execute approx. 74M instructions
  for (volatile int i = 0; i < 12345678; i++)
    ;

  float t1 = time_in_secs();

  printf("%f secs elapsed\n", t1 - t0);

  return 0;
}
$ riscv64-unknown-elf-gcc -O2 test.c
$ spike pk ./a.out
bbl loader
0.074074 secs elapsed
$ 
```

scottj97, 2021-11-27:
> I just want to measure the time taken between some statements of code.

On what RISC-V implementation? Your goal is incomplete. It's like asking, "How fast is the car going when the engine is at 2000 rpm?" The answer is: it depends!

If you're trying to measure the performance of Spike itself (why?) then the real-time CLINT might be useful, or you could use a hosted profiler like [gprof](https://ftp.gnu.org/old-gnu/Manuals/gprof-2.9.1/html_node/gprof_1.html).

If you're trying to measure the performance of a specific RISC-V silicon implementation, Spike can only provide an estimate, as I explained above, based on instruction count. Andrew's answer is a nice way to get that estimate, assuming 1GHz and an IPC of 1.00, which may or may not match any actual implementation. The IPC can also vary widely based on the workload, and Spike, not being a performance model, will not account for that, which means your estimate might be way off.

You might get more help in a [RISC-V software group](https://groups.google.com/a/groups.riscv.org/g/sw-dev).