riscv-software-src/riscv-isa-sim #862: Exceptions interrupted when delegating

sammy17, 2021-11-18:
I wanted to change Spike to align with the hardware deisgn I am testing (BOOM) and for that I changed the allowed mdeleg exceptions in spike to match those of BOOM by changing the following statement:

https://github.com/riscv-software-src/riscv-isa-sim/blob/37b7204920e1cf5adbdc6241fb20192c3a2ee495/riscv/csrs.cc#L732

I changed the statement to below:
```
bool medeleg_csr_t::unlogged_write(const reg_t val) noexcept {
  const reg_t mask = 0
    |   (1 << CAUSE_MISALIGNED_FETCH) |
        (1 << CAUSE_ILLEGAL_INSTRUCTION) |
        (1 << CAUSE_BREAKPOINT) |
        (1 << CAUSE_MISALIGNED_LOAD) |
        (1 << CAUSE_MISALIGNED_STORE) |
        (1 << CAUSE_USER_ECALL) |
        (1 << CAUSE_FETCH_PAGE_FAULT) |
        (1 << CAUSE_LOAD_PAGE_FAULT) |
        (1 << CAUSE_STORE_PAGE_FAULT)
    | (proc->extension_enabled('H') ? hypervisor_exceptions : 0)
    ;
  return basic_csr_t::unlogged_write((read() & ~mask) | (val & mask));
}
```

After the change, spike allows setting medeleg for these exceptions but when exceptions are delegated it gives an **exception interrupt #7** when trying to execute the delegated exception handler and jump to the mmode exception handler. But in this case BOOM jumps to the delegated exception handler and continues execution without any issue. 

This is the code snippet that trigger the issue, I have attached the binary file for the program as well. 
```
lui	t0,0xb
addiw	t0,t0,265
csrs	medeleg,t0 # medeleg set to 0xb109

auipc	t0,0x0
addi	t0,t0,-668 # 8000012c <csr_dump>
csrw	mtvec,t0   #mtvec is set to csr_dump

auipc	t0,0x0
addi	t0,t0,-740 # 80000114 <_suffix>
csrw	stvec,t0    # stvec is set to _suffix

# program jumps to user mode and continue execution
# On ecall I receive following exceptions in spike
core   0: 0x0000000080000510 (0x00000073) ecall
core   0: exception trap_user_ecall, epc 0x0000000080000510
core   0: exception interrupt #7, epc 0x0000000080000114
core   0: >>>>  csr_dump
# Jumps to csr_dump and continue
```
First instruction of _suffix is **fcvt.wu.s	t4,fa6,rne** so it has nothing to do with 7 - Store/AMO access fault from what I see. 

Could you please let me know whether the changes I did are sufficient to change the delegation behaviour? If so, what is the reason that I am seeing this exception interrupts? 

Attached:
[test.elf.zip](https://github.com/riscv-software-src/riscv-isa-sim/files/7565678/test.elf.zip)

scottj97, 2021-11-18:
I think your change is sufficient. But from what you posted above (I didn't look at your attachment) I can't see that you're using the modified behavior, since user-mode ecall is allowed to be designated both before and after your change. Does this behavior change when you revert your modification?

Interrupt #7 is a machine-mode timer interrupt, which could be legit, and likely has nothing to do with your change.

sammy17, 2021-11-19:
Thanks for the comment, I just verified that the same behavior can be observed with the unchanged Spike as well. I observed the same when other exceptions are being delegated, right at that moment I see interrupt #7 triggered. That's why I thought it might be related to the change I did. 

Do you know why machine timer interrupt is being triggered right at that moment? It does not seem like a coincidence/random since I see the same in a few other tests. 
And also can you point me the place to disable machine timer interrupt in spike, I am not planning to test interrupt behavior for now. 
Thank you in advance!

aswaterman, 2021-11-19:
Spike samples the interrupts both periodically and at specific synchronization points, including when other traps are taken.  Most likely, the interrupt had been pending for a little while, but the act of taking it was triggered by taking the trap.

sammy17, 2021-11-19:
Thanks, that makes sense. Is there any way to disable timer interrupts internally in spike without changing/hardcoding any CSR values?

sammy17, 2021-11-19:
I was able to disable all interrupts by commenting https://github.com/riscv-software-src/riscv-isa-sim/blob/37b7204920e1cf5adbdc6241fb20192c3a2ee495/riscv/processor.cc#L727 and https://github.com/riscv-software-src/riscv-isa-sim/blob/37b7204920e1cf5adbdc6241fb20192c3a2ee495/riscv/processor.cc#L729 for now.

closed 2021-11-19 by sammy17