riscv-software-src/riscv-isa-sim #864: delegable_ints excludes machine interrupts seemingly arbitrarily

p0llard, 2021-11-25:
https://github.com/riscv-software-src/riscv-isa-sim/blob/37b7204920e1cf5adbdc6241fb20192c3a2ee495/riscv/csrs.cc#L608

From my reading of 20190608-Priv-MSU-Ratified, there doesn't seem to be anything that would _forbid_ Machine Interrupts from being delegated, so it should be possible to set `mideleg[i]` for `i \in {11, 7, 3}`; of course the implementation is allowed not to support this.

It seems that `MIP_MSIP | MIP_MTIP | MIP_MEIP` are only ORed into `all_ints` and have thus been consciously excluded from `delegable_ints`.

It seems that either:
1. my reading of the standard is wrong
2. there's a good reason why an implementation might want to forbid delegation of Machine Interrupts, and Spike has chosen to follow this

It seems the fu740 doesn't support this delegation, so I'm leaning towards 2, but I can't really see why this is.

aswaterman, 2021-11-25:
The reasoning goes the other way: there isn't a good reason to delegate those interrupts, and so Spike takes the same minimalist approach we expect the majority of HW implementations to take.  In practice, this simplifies verification/cosimulation a little bit.

p0llard, 2021-11-25:
@aswaterman fair enough, closing since this clearly isn't an oversight.

For context, I want to delegate the Machine Timer Interrupt to Supervisor mode; it seems that most silicon doesn't break out a Supervisor Timer line, so it seems that if one wants the Supervisor to receive Timer interrupts then one must "soft delegate" by having the `mtvec` raise `STIP`.

It seems that it would be more efficient to have the hardware delegate here, especially in the case where there's nothing actually running in Machine mode (i.e. no SBI)?

closed 2021-11-25 by p0llard

aswaterman, 2021-11-26:
The recently ratified Sstc extension offers a directly usable supervisor timer interrupt.  The approach is a little different than what you're suggesting, because it adds a second comparator, rather than stealing M-mode's.