riscv-software-src/riscv-isa-sim #865: Exception mismatch between Spike and Rocket

sammy17, 2021-11-25:
I have a testcase where I observe two different exceptions on the same PC in Spike Vs Rocket (and BOOM). 

I have attached the binary but this is the configuration summary in the test:
```
stvec     = 0xffffffffffe00330
sepc      = 0x0000000000002818
medeleg   = 0x000000000000b100 (I have changed exceptions that are allowed to delegated in spike to match HW)
pmpaddr0  = 0x000000003fffffff  (I have set #define MAX_PADDR_BITS 32 in spike to match HW)
pmpcfg0   = 0x000000000000001f 
satp      = 0x8000000000080006
```

Test setup these CSRs in machine mode and execute SRET, and as expected program jumps to  SEPC =**0x0000000000002818** in umode which trigger an instruction page fault exception. 
Since this exception is delegated, the program jumps to stvec (0x000000003fffffff). 

**Up to this point, Spike and Rocket behaviors are similar.** 

At PC=stvec Spike triggers an illegal instruction as shown below while Rocket gives an instruction access fault. 
From the looks of it, Spike fetches all zeros from the address and considers it as an illegal instruction while Rocket gives an exception due to access permission failure (I am trying to confirm this).

```
core   0: 0x000000008000032c (0x10200073) sret
core   0: 3 0x000000008000032c (0x10200073) c1536_hstatus 0x0000000200000000 c768_mstatus 0x8000000a00006020
core   0: exception trap_instruction_page_fault, epc 0x0000000000002818
core   0:           tval 0x0000000000002818
core   0: 0xffffffffffe00330 (0x00000000) c.addi4spn s0, sp, 0
core   0: exception trap_illegal_instruction, epc 0xffffffffffe00330
core   0:           tval 0x0000000000000000
core   0: >>>>  csr_dump
core   0: 0x00000000800001e0 (0x00003097) auipc   ra, 0x3
```

 **Could you please let me know if there are any known reasons for this mismatch?** 
Any pointers to debug this issue further is welcome. 

Attached Spike log and binary:
[spike_1.log](https://github.com/riscv-software-src/riscv-isa-sim/files/7605334/spike_1.log)
[test.zip](https://github.com/riscv-software-src/riscv-isa-sim/files/7605335/test.zip)

aswaterman, 2021-11-26:
Most likely these are both legal behaviors, based upon different system configurations and ISA flexibility.  We wont’t investigate.  Let us know if you figure it out.

aswaterman, 2021-11-27:
I did try to look into this after all, but your `test.elf` doesn't match your `spike_1.log`.  They diverge when the PC hits 0x80000298: the ELF has `andi    sp, sp, -129`, but the log has `li gp, 1`.

Also, when I execute the ELF on Spike, I do get an access exception after `sret` to 0x2818, as you reported for Rocket.

sammy17, 2021-11-27:
Thank you very much for your time. I am sorry, there might have been a mixup with the log. 
But ELF is correct and I verified what you said with the latest spike as well. 
So latest spike gives access fault after sret to 0x2818 but my edited version of spike with the changes that I mentioned above result in a different instruction sequence. 
Changes that I did are related to PMP, medeleg and initial value of mstatus=0xa00001800 to match BOOM.

I have attached the correct logs with the latest spike and my edited version of the spike. I will summarize the differences below as well. 

**Latest spike (unedited):**
```
core   0: 0x00000000800004a8 (0x10200073) sret
core   0: 3 0x00000000800004a8 (0x10200073) c1536_hstatus 0x0000000200000000 c768_mstatus 0x8000000a00006020
core   0: exception trap_instruction_access_fault, epc 0x0000000000002818
core   0:           tval 0x0000000000002818
core   0: >>>>  csr_dump
core   0: 0x00000000800001e0 (0x00003097) auipc   ra, 0x3
core   0: 3 0x00000000800001e0 (0x00003097) x 1 0x00000000800031e0
```
**Edited spike:**
```
core   0: 0x00000000800004a8 (0x10200073) sret
core   0: 3 0x00000000800004a8 (0x10200073) c1536_hstatus 0x0000000200000000 c768_mstatus 0x8000000a00006020
core   0: exception trap_instruction_page_fault, epc 0x0000000000002818
core   0:           tval 0x0000000000002818
core   0: 0xffffffffffe004ac (0x00000000) c.addi4spn s0, sp, 0
core   0: exception trap_illegal_instruction, epc 0xffffffffffe004ac
core   0:           tval 0x0000000000000000
core   0: >>>>  csr_dump
core   0: 0x00000000800001e0 (0x00003097) auipc   ra, 0x3
core   0: 3 0x00000000800001e0 (0x00003097) x 1 0x00000000800031e0
```

Logs:
[spike_edited.log](https://github.com/riscv-software-src/riscv-isa-sim/files/7611306/spike_edited.log)
[spike_latest.log](https://github.com/riscv-software-src/riscv-isa-sim/files/7611307/spike_latest.log)

I tried to debug the issue further and so far what I got is edited Spike version translated virtual address 0xffffffffffe004ac to physical address 0xbfe004ac without triggering any access fault and instruction is fetched as all zeros.

aswaterman, 2021-11-27:
I missed your remark about `MAX_PADDR_BITS`.  That isn't meant to be a configuration parameter.  We don't guarantee that Spike will behave correctly if you make any changes to the C code.

aswaterman, 2021-11-27:
I do have an explanation for you, though.  You're programming `satp.PPN` to `480006`, which corresponds to physical address `480006000`.  Obviously, that physical address is illegal if you don't have a huge amount of DRAM, and so raising an access exception is correct.

But reducing `MAX_PADDR_BITS` reduces the width of the `satp.PPN` field.  If `satp.PPN` is narrowed to only hold addresses up to 2^32, writing `480006` has the same effect as writing `80006`.  But address `80006000` is actually legal, so there's no access exception.  And the PTE at that address is 0, so a page fault exception is correctly raised.  Since that exception is delegated, you jump to `stvec`.  Since `stvec` ultimately maps to physical address `bfe004ac`, which contains the value 0, raising an illegal instruction is correct.

So, even though messing with the C code isn't warrantied behavior, it looks like Spike is doing something rational in both cases.

sammy17, 2021-11-27:
I see, thank you very much for the explanation. It is clear now. I double-checked the PMP configuration used in this test as well, I don't see a reason why Rocket/BOOM is giving access fault on paddr=bfe004ac. 

Since pmpaddr0=0x3fffffff and pmpcfg0=0x1f, all 32bit physical addresses should be valid.  
Address **0xbfe004ac is not initialized/defined within the program**, I wonder whether that makes a difference. 
I will debug this further from the HW side and see why Rocket/BOOM is giving access fault for this address. 

I will close this issue since this mismatch is only seen in the edited version of Spike. But comments are always welcome. 
Thank you again for all the help.

closed 2021-11-27 by sammy17

aswaterman, 2025-12-09:
You're doing it right already.  If you don't have memory mapped at the high addresses, then accessing the high addresses will raise an exception.  That really is no different than reducing the physical-address width.

omerguzelelectronicguy, 2025-12-10:
> > I missed your remark about `MAX_PADDR_BITS`. That isn't meant to be a configuration parameter. We don't guarantee that Spike will behave correctly if you make any changes to the C code.
> 
> [@aswaterman](https://github.com/aswaterman) Is there any mechanism to set narrower max paddr bits? It was working fine but when I pull the latest version I saw that it is removed. Is there a mechanism spike to behave the rtl design(pmpaddr masking)?
> 
> edit:
> 
> ```
> 	memory@A0000000 {
> 		device_type = "memory";
> 		reg = <0x00 0xa0000000 0x00 0x02000000>;
> 	};
> ```
> 
> this is from my dtb file. It seems dtb doesn't help me to limit the maximum physical address.

For the ones who has same goal can use `paddr_bits` function in `processor.h` . Since this is changing the c code too, there is no guarantee that Spike correctly works.