riscv-software-src/riscv-isa-sim #872: Trigger on HLV should set mstatus.GVA=1

scottj97, 2021-12-04:
In #850 I fixed the `mstatus.GVA` bit for normal loads & stores but didn't consider HLV/HLVX/HSV or load/store with MPRV=1 and MPV=1. These types of load/stores today will mistakenly set GVA=0.

scottj97, 2021-12-05:
To solve this robustly, I am thinking of a data structure that holds info about the desired access:
* virtual address
* effective mode (including 2 bits of privilege and 1 bit of V)
* special cases that alter permissions like HLVX

Then we pass around this data structure instead of a simple virtual address.

This would replace things like `xlate_flags` with its `RISCV_XLATE_VIRT` and `RISCV_XLATE_VIRT_HLVX`.

aswaterman, 2021-12-09:
That sounds like a sound and more robust approach to me, Scott.

rbuchner-aril, 2023-10-09:
@aswaterman I believe this should be closed.

closed 2023-10-09 by aswaterman