riscv-software-src/riscv-isa-sim #878: XS field in mstatus/status is read-only, but sstatus_write_mask allows to write

Lemover, 2021-12-07:
In the privileged spec:
  
The FS[1:0] WARL field and the XS[1:0] read-only field are used to reduce the cost of context save and restore by setting and tracking the current state of the floating-point unit and any other usermode extensions respectively.

**XS field should be read-only.**
 When I look for spike's relative implementation:
```c++
reg_t base_status_csr_t::compute_sstatus_write_mask() const noexcept {
  // If a configuration has FS bits, they will always be accessible no
  // matter the state of misa.
  const bool has_fs = proc->extension_enabled('S') || proc->extension_enabled('F')
              || proc->extension_enabled_const('V');
  const bool has_vs = proc->extension_enabled_const('V');
  return 0
    | (proc->extension_enabled('S') ? (SSTATUS_SIE | SSTATUS_SPIE | SSTATUS_SPP) : 0)
    | (has_page ? (SSTATUS_SUM | SSTATUS_MXR) : 0)
    | (has_fs ? SSTATUS_FS : 0)
    | (proc->any_custom_extensions() ? SSTATUS_XS : 0)
    | (has_vs ? SSTATUS_VS : 0)
    ;
}
```
link:
https://github.com/riscv-software-src/riscv-isa-sim/blob/9f18486669df4eddf0e3d799d686a9714d348bc8/riscv/csrs.cc#L339

It looks like that if there are any custom extendsions, XS field will be writable.
Is there anything that I ignored?

scottj97, 2021-12-07:
I suspect you are correct. I'm not sure if this ever worked correctly. I rewrote all this logic recently but from what I can tell XS has always been writable if any custom extensions are defined. Perhaps the spec for XS changed but Spike was never updated?

aswaterman, 2021-12-07:
Not sure of the history, but I agree this is not what the spec intends. But it’s a bit more complicated to fix than simply making the field read-only, since we need to expose an API to the custom-extension stuff.

closed 2021-12-07 by Lemover