riscv-software-src/riscv-isa-sim #902: mstatus.FS is not set to dirty when mstatus.FS is changed

sammy17, 2022-01-11:
I noticed that old spike version I have (I do not have the commit version readily) set the mstatus.FS to dirty when mstatus.FS is changed from 2'b00 to 2'b01 as below example:

```
a0 = 0x8000000a00007800
mstatus = 0x0000000000003800 (FS=2'b01)
    csrs    mstatus, a0
In old spike   : mstatus = 0x8000000a00007800 (FS=2'b11)
In new spike : mstatus = 0xa00003800 (FS=2'b01)
```

**Could you please explain the reason for this difference between old and new spike?** 

I thought setting the FS to dirty in the old spike is justified by the following table in Table 3.4 on Privilege spec (20190608).
[image]

aswaterman, 2022-01-11:
FS isn't considered part of the FP state, and so the row in the table is not relevant.

It turns out the old behavior and the new behavior are both legal.  The new behavior is more precise (FS is only set to Dirty when necessary, and so avoids some unnecessary saves and restores).  The old behavior is sloppy, but still legal (https://github.com/riscv/riscv-isa-manual/blob/76bd881ae086983589f2c70c32bf8163afb6e363/src/machine.tex#L1028-L1030).

sammy17, 2022-01-11:
Thank you very much for the explanation, it is clear now. 

I have another similar scenario to get a confirmation. I see that spike does not set FS to dirty when writing to fcsr (when the value of fcsr is not changed by the write) but BOOM set it to dirty in this scenario. In this case is BOOM's implementation 'sloppy' i.e. does it reduces performance in BOOM by doing this? 

```
fcsr = 0x0
 csrwi   fcsr, 0
FS is set to dirty in BOOM
FS is unchanged in latest spike
```
**According to the document that you referred it is clear that both these scenarios does not violate ISA spec but I wanted to know whether one is better than the other in terms of performance as you mentioned for the other scenario.**

scottj97, 2022-01-11:
You're right, BOOM is unnecessarily setting Dirty here. Any unnecessary setting of Dirty like this may reduce performance, but in this instance I'd guess the impact is negligible.

Strangely, from looking at the code, I don't see the Spike behavior you observe. I'd expect any `csrwi` to `fcsr` to set FS=Dirty.

It also seems unreasonable in any hardware implementation to compare the old and the new value and decide to Dirty based on that. In a straightforward implementation, any write to `fcsr` will set Dirty.

sammy17, 2022-01-12:
Thank you very much for the confirmation.

That is strange, my version of spike has the above mentioned behaviour, I will let you know the commit version in a bit. I have made some changes in spike to match HW implementation but I don't think it affects FS behaviour. 

I thought above mentioned behaviour in new spike is justified by this [sentence](https://github.com/riscv/riscv-isa-manual/blob/76bd881ae086983589f2c70c32bf8163afb6e363/src/machine.tex#L1037).

sammy17, 2022-03-04:
Yeah, it can be closed. Thank you for the help!

closed 2022-03-04 by sammy17