riscv-software-src/riscv-isa-sim #915: [question] mtval update on consecutive exceptions

mysoreanoop, 2022-01-30:
When encountering consecutive exceptions, specifically:
1.  illegal instruction exception, and then
2.  instruction access fault exception,
spike seems to be writing to mtval, the information about the exception 1 instead of 2 (like Dromajo). It is not clear, from the spec, which is the right way; but seems like the latest exception information has to be written. Thoughts?

For example, here's a test case where the above situation occurs in Dromajo: 
[image]
In the figure, the 2nd column is the current privilege mode, sequentially executed PCs are highlighted like so:
green -- sret instruction
yellow -- illegal instruction exception on return address
pink -- instruction access fault on (supervisor?) trap handler
orange -- (user mode?) trap handler
Later on, when mtval is read as part of the handler, dromajo returns: `0xffffffffffe0033c`
Also note that the first exception information in mtval should be the illegal instruction bits itself, instead of the PC.

The same program executed on spike leads to:
[image]
Later on, when mtval is read as part of the handler, spike returns: ` 0x0000000000000bfc`

mysoreanoop, 2022-01-30:
[tests.zip](https://github.com/riscv-software-src/riscv-isa-sim/files/7965710/tests.zip)
Here's the full test case described above.

neelgala, 2022-01-31:
spike gives you an access fault instead of the illegal fault that dromajo gives. This happens because satp.mode is set to 0 which is no-translation mode. So 0xbfc is the machine address. But in spike the accessible region is only 0x8000_0000 and above, hence an access fault in spike, for which the `mtval` value is correct.

mysoreanoop, 2022-01-31:
Thank you very much, that makes sense.

closed 2022-01-31 by mysoreanoop