riscv-software-src/riscv-isa-sim #931: SIGSEGV Segmentation fault occurring in Spike when using custom extension

donreed-cornelisnetworks, 2022-02-25:
I recently up leveled my Spike repo from Sept. 17th, 2021, to being current with Master on Feb. 24th.  At that time, I began experiencing segfaults terminating Spike when using custom extensions - occurring at ../riscv/rocc.cc:27

I traced the problem to a commit on January 26th which removed the code that inserted custom extensions into the custom_extension container.  Specifically the following lines were removed:

   if (!custom_extensions.insert(std::make_pair(x->name(), x)).second) {
      fprintf(stderr, "extensions must have unique names (got two named \"%s\"!)\n", x->name());
      abort();
   }

This causes the extension to be missing from the processor object resulting in the segfault.

scottj97, 2022-02-25:
I suppose you're talking about d685520fed95e6518d6c2174ae51310292b891fe which did not remove the code in question, but moved it from one place to another. Part of #912. @liweiwei90 ?

donreed-cornelisnetworks, 2022-02-25:
Yes that is the commit.

liweiwei90, 2022-02-26:
> I suppose you're talking about [d685520](https://github.com/riscv-software-src/riscv-isa-sim/commit/d685520fed95e6518d6c2174ae51310292b891fe) which did not remove the code in question, but moved it from [one place](https://github.com/riscv-software-src/riscv-isa-sim/commit/d685520fed95e6518d6c2174ae51310292b891fe#diff-96f446fa4c024af925fb82998baa4145356afb5a6940d639fa705162029c8a78L1102-L1105) to [another](https://github.com/riscv-software-src/riscv-isa-sim/commit/d685520fed95e6518d6c2174ae51310292b891fe#diff-96f446fa4c024af925fb82998baa4145356afb5a6940d639fa705162029c8a78R347-R350). Part of #912. @liweiwei90 ?

Yes. I have moved the code because I found the disasm for custom extensions may be registered duplicated in  register_extension(...) and processer(...). 
I think the difference caused by my changes is that the extensions added by "--extension" will not be inserted to custom_extension which seems have no effect on current functions.
When I tested " ./spike --isa=rv64gc --extension=dummy_rocc" command, it seems truely trigger segfault in register_extension()  both with and without my change. I'll try to find the problem later.

liweiwei90, 2022-02-26:
I found 2 problem:
1.  the insn_desc_t structure have been changed(add support for rv32e/rv64e). However the code in riscv/rocc.cc and customext/clflush.cc doesn't change:
   ` {0x0b, 0x7f, &::illegal_instruction, c0}`
However I'm a liitle confused about the result of my modification:
a. When I change the struct to the original small struct as follows. It can work well. 
```
struct insn_desc_t1
{
  insn_bits_t match;
  insn_bits_t mask;
  insn_func_t rv32i;
  insn_func_t rv64i;
};

std::vector<insn_desc_t1> rocc_t::get_instructions()
{
  std::vector<insn_desc_t1> insns;
  insns.push_back((insn_desc_t1){0x0b, 0x7f, &::illegal_instruction, c0});
  insns.push_back((insn_desc_t1){0x2b, 0x7f, &::illegal_instruction, c1});
  insns.push_back((insn_desc_t1){0x5b, 0x7f, &::illegal_instruction, c2});
  insns.push_back((insn_desc_t1){0x7b, 0x7f, &::illegal_instruction, c3});
  return insns;
}
```
   
b. when I adapt get_instructions to fit into current structure such as:

```
std::vector<insn_desc_t> rocc_t::get_instructions()
{
  std::vector<insn_desc_t> insns;
  insns.push_back((insn_desc_t){0x0b, 0x7f, &::illegal_instruction, c0, &::illegal_instruction, c0});
  insns.push_back((insn_desc_t){0x2b, 0x7f, &::illegal_instruction, c1, &::illegal_instruction, c0});
  insns.push_back((insn_desc_t){0x5b, 0x7f, &::illegal_instruction, c2, &::illegal_instruction, c0});
  insns.push_back((insn_desc_t){0x7b, 0x7f, &::illegal_instruction, c3, &::illegal_instruction, c0});
  return insns;
}
```
 
 The size of insns vector returned by get_instructions() is very large and will trigger segfault at last. 
It seems the insns is destroyed when insn_desc_t structure become larger. I don't know why?

2. the second problem is the difference introduced by my change: the functions in ROCC will try to find extension from  custom_extension:
    ` rocc_t* rocc = static_cast<rocc_t*>(p->get_extension());`
which will return NULL is ROCC extension is specified by "--extension"

liweiwei90, 2022-02-26:
the fix for second problem can be found in https://github.com/plctlab/plct-spike/tree/plct-segfault-fix

liweiwei90, 2022-03-02:
> 1. However I'm a liitle confused about the result of my modification:

This is caused by a problem in my local environment which make the new libcustomext.so didn't work.
The first problem seems have no relationship with segfault.

I also have fixed it by the way.

aswaterman, 2022-03-02:
I've merged #937.

closed 2022-03-07 by donreed-cornelisnetworks