riscv-software-src/riscv-isa-sim #936: new feature： spike simmulation for multi core with given number of instructions for each core per time interval

gxflying, 2022-02-28:
Hi sir, as I know in in the spike , the instructions to be executed for each core per time interval is defined in the sim.h， it is statical number and can not be dynamicly changed when run the simulation.   
https://github.com/riscv-software-src/riscv-isa-sim/blob/fd28dd0ecfaea0f7ca15183303c79159b155776c/riscv/sim.h#L104
but I think it a important featrue when think about multi core senario, especially when we use spike as a golden model to generate the dynamic stream of a program which may be compared with a stream (trace) genterated by rtl.  
if the spike can control the "INTERLEAVE" which is the instruction to be  run on a certain core dymanicly , it would be more friendly for the multi-core veridication.  
for example,  the value of "INTERLEAVE" can be from a file with information below, including some timing information .
hart0 executes 100 instrustions, then hart2 110 instructions, then hart1 50 instructhons, .... and so on according rtl trace .
by default the  "INTERLEAVE" can be set to the same value when no need to be compared with rtl trace.  
hartid       instructions executed
0               100
2                110
1                 50
3                 200
3                 40
0                  80
.....

do we consider about this feature and  have plan to add it into spike ?

scottj97, 2022-02-28:
I use Spike as a golden model for comparison against a multi-hart RTL design. It requires so much more than just this simple feature. Unpredictable values such as `mcycle` need to be fed to Spike to maintain sync. External interrupts, whose timing is not predictable to Spike, need to be fed from RTL. Load reservations might have behavior unpredictable by Spike, causing the behavior of `SC` to differ. The interleaving of the two cores is just one small thing of all the things that need to be synchronized.

On top of all that, what happens when two or more harts commit instructions simultaneously? Spike cannot model that.

gxflying, 2022-03-01:
thanks  for your explanation.
yes，there are morethings to be done to support multi-hart comparison
but the issues you mentioned like mcycle/ interrupt actually need to  be  handled not only for multicore but also for single core scenario
about the  lr/sc the how to decide the reservation sack will be invalid is a  implementation specified behavior anyway
about the top if all issue that commiting  instruction simultaneously,  actually I think it is not a issue at all,
because in multicore , 
only load/ store need to be taken more attentions, and load/ store will NEVER commit at the same time when operating on the same / overlaped address because of the mem coherence of multicore RTL
 other type instructions and load/store on differnt address the new featrue I mentioned above is enough to ensure the right result
 in all I think the new feature is a coomon featrue and is useful for most multicore scenarios 
how do you think about ?

scottj97, 2022-03-01:
I don't agree that it's useful. For comparison vs RTL you want fine-grained control of Spike, stepping it one instruction at a time, enabling feedback from RTL sim into Spike at every instruction boundary. You can use that same API to control stepping of each hart.

(And just to be pedantic, non-coherent stores can commit simultaneously to the same address.)

gxflying, 2022-03-01:
non-coherent stores is a system level or soc level question, I don't think can be resovled in a instruction set simulator,  if  store into io space , the store will have not influence on the instruction stream after it,    if store into shared mem space without non-coherent , it always need some instruction to maintain conherence between masters(cpu/gpu..)  explicitly in rtl simulation
the new feature will make the multicore trace of spike just like a single core when comparision against a RTL design
anyway, if this feature seems to be not so useful to all other people who use spike in multicore senario by now
maybe I can implement this feature in my own branch.
by the way, using the "step" api is a alternative
thanks

closed 2022-03-01 by gxflying