riscv-software-src/riscv-isa-sim #941: set_vl function seems to incorrectly compute vill for fractional LMUL

amorenosmd, 2022-03-07:
Hi,

It seems that when running code that sets a fractional LMUL with SEW=ELEN Spike sets VILL to 1. Looking at the code I saw the following condition within the set_vl function (processor.cc):

`
    vill = !(vflmul >= 0.125 && vflmul <= 8)
           || vsew > std::min(vflmul, 1.0f) * ELEN
           || (newType >> 8) != 0;
`

I'm a bit surprised in particular by the condition:
`
vsew > std::min(vflmul, 1.0f) * ELEN
`

Which basically suggests that any SEW that matches ELEN can never set a fractional LMUL. I believe this may be a bug and the correct condition should probably use with "VLEN" instead of "ELEN".

aswaterman, 2022-03-08:
This is correct behavior.  The spec says, "For a given supported fractional LMUL setting, implementations must support SEW settings between SEWMIN and LMUL * ELEN, inclusive."

(This is also consistent with the main purpose of fractional LMUL, which is to reduce register pressure when handling multiple widths.  LMUL=1/2 only saves register pressure vs. LMUL=1 for instructions that produce outputs wider than SEW.  But if SEW=ELEN, there are no such instructions.  So LMUL=1/2 SEW=ELEN doesn't really buy you anything.)

amorenosmd, 2022-03-08:
Thanks for the quick answer and the clarification. I missed that line in the spec :)

closed 2022-03-08 by amorenosmd