riscv-software-src/riscv-isa-sim #943: Can't run spike hello world program ,An illegal instruction was executed!

kkmonk, 2022-03-09:
both spike and pk are built with 64bit
and riscv-gcc toolchain is also 64bit
hello world progrm
```c
#include <stdio.h>
int main()
{
    int a= 1;
    a = a+2;
    printf("%d\n",a);
    return 0;
}
```
gcc cmd : `riscv64-unknown-elf-gcc demo.c -o demo` 
spike cmd: `spike pk demo`

spike output:
```
[xuelai@X86 c_test]$ spike pk demo
bbl loader
z  0000000000000000 ra 0000000000000000 sp 0000000000000000 gp 0000000000000000
tp 0000000000000000 t0 000000000001e600 t1 0000000000000000 t2 0000000000000000
s0 0000000000000000 s1 0000000000000000 a0 0000000000000000 a1 0000000000000000
a2 0000000000000000 a3 0000000000000000 a4 0000000000000000 a5 0000000000000000
a6 0000000000000000 a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 0000000000000000 t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc 0000000000010108 va/inst 000000003002a073 sr 8000000200006020
An illegal instruction was executed!
```

and i disassemble the `demo` bin file and search the illegal instruction,which is
```
[xuelai@X86 c_test]$ riscv64-unknown-elf-objdump -d demo|grep 3002a073
   10108:       3002a073                csrs    mstatus,t0
   1024c:       3002a073                csrs    mstatus,t0
```
here is my working env:
```
[xuelai@X86 c_test]$ uname -a
Linux X86 3.10.0-1160.25.1.el7.x86_64 #1 SMP Wed Apr 28 21:49:45 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

scottj97, 2022-03-11:
I don't see anything obviously wrong in your flow. Perhaps you need to specify a 64-bit ISA on the command-line? e.g `spike --isa=rv64g pk demo`

Wild guess, though.

scottj97, 2022-03-11:
Also it would be helpful if you could turn on `-l --log-commits` and paste here the commit log around the illegal instruction exception.

scottj97, 2022-03-18:
Closing since OP has not responded.

closed 2022-03-18 by scottj97