riscv-software-src/riscv-isa-sim #947: How to view the histogram created by `-g`? (after configuring/building spike with the `--enable-histogram` flag)

gkielian, 2022-03-14:
Was able to get both the `-l` log and `--log-commits` no problem, saving output via `--log=test.log`.

But now I'm adding in the `--enable-histogram`, but not sure how to view the PC Histogram. 

After running spike like before but adding  `-g` flag, the logs return as normal, but haven't been able to find example or where to look for the output. (Tried to thoroughly check normal channels, issues, and help menus for this, but no luck).

Would this be accessible via an interactive command? or perhaps should it just output to stdout if successful?

Thanks for you help!
Greg

scottj97, 2022-03-14:
I've also tried to use the histogram in the past and could not figure out how to make it do anything at all.

aswaterman, 2022-03-14:
Just tested; it does work.  Configure with `--enable-histogram` and pass the `-g` flag.  The histogram is printed to stderr.

gkielian, 2022-03-14:
Thanks @aswaterman -- knowing that it was supposed to go to stderr really helped narrow it down, and I was able to figure it out.

Took a closer look at my output trace and saw the following line hidden in there; looks like the proxy kernel is needed in order to get the histogram.

```sh
warning: tohost and fromhost symbols not in ELF; can't communicate with target
```
## Solution: pc opcode histogram option requires invocation of the proxy kernel

After modifying the command from `spike -g test.elf` to:

`spike -g ./path/to/pk test.elf`

Then histogram was produced and sent to stderr, and looks like this:

```
  PC Histogram size:3755
  1000 1
  1004 1
  1008 2
  100c 1
  1010 1
  30000000 1
  30000008 1
  3000000c 1
  30000010 1
  30000014 1
  ...
 ```

Closing comment, and hoping this helps others having the same issue.

closed 2022-03-14 by gkielian

aswaterman, 2022-03-14:
Oh, interesting, I wasn't aware of that idiosyncrasy.  Thanks for posting this for posterity.

farukyld, 2025-01-15:
does this mean that we can't see the pc histogram if we run spike with opensbi as `<target program>` and linux kernel Image as `--kernel` and with an `--initrd`?

maybe we can attach the spike process with gdb and inspect the process' internal variables.