riscv-software-src/riscv-isa-sim #952: How to enable support for misaligned loads/stores

aignacio, 2022-03-18:
Hey,

I'm trying to enable misaligned loads/stores in the spike build but even if I configure the correct argument with `../configure --enable-misaligned` it still shows me this:
```bash
$ ./spike --enable-misaligned
./spike: unrecognized option --enable-misaligned                                                                                                                           │70956 aignacio   17   0 57.8G 66140 ?  0.
Try 'spike --help' for more information.
$ ./spike -h
spike -h
Spike RISC-V ISA Simulator 1.1.1-dev

usage: spike [host options] <target program> [target options]
Host Options:
  -p<n>                 Simulate <n> processors [default 1]
  -m<n>                 Provide <n> MiB of target memory [default 2048]
  -m<a:m,b:n,...>       Provide memory regions of size m and n bytes
                          at base addresses a and b (with 4 KiB alignment)
  -d                    Interactive debug mode
  -g                    Track histogram of PCs
  -l                    Generate a log of execution
  -h, --help            Print this help message
  -H                    Start halted, allowing a debugger to connect
  --log=<name>          File name for option -l
  --debug-cmd=<name>    Read commands from file (use with -d)
  --isa=<name>          RISC-V ISA string [default RV64IMAFDC]
  --priv=<m|mu|msu>     RISC-V privilege modes supported [default MSU]
  --varch=<name>        RISC-V Vector uArch string [default vlen:128,elen:64]
  --pc=<address>        Override ELF entry point
  --hartids=<a,b,...>   Explicitly specify hartids, default is 0,1,...
  --ic=<S>:<W>:<B>      Instantiate a cache model with S sets,
  --dc=<S>:<W>:<B>        W ways, and B-byte blocks (with S and
  --l2=<S>:<W>:<B>        B both powers of 2).
  --device=<P,B,A>      Attach MMIO plugin device from an --extlib library
                          P -- Name of the MMIO plugin
                          B -- Base memory address of the device
                          A -- String arguments to pass to the plugin
                          This flag can be used multiple times.
                          The extlib flag for the library must come first.
  --log-cache-miss      Generate a log of cache miss
  --extension=<name>    Specify RoCC Extension
                          This flag can be used multiple times.
  --extlib=<name>       Shared library to load
                        This flag can be used multiple times.
  --rbb-port=<port>     Listen on <port> for remote bitbang connection
  --dump-dts            Print device tree string and exit
  --disable-dtb         Don't write the device tree blob into memory
  --kernel=<path>       Load kernel flat image into memory
  --initrd=<path>       Load kernel initrd into memory
  --bootargs=<args>     Provide custom bootargs for kernel [default: console=hvc0 earlycon=sbi]
  --real-time-clint     Increment clint time at real-time rate
  --dm-progsize=<words> Progsize for the debug module [default 2]
  --dm-sba=<bits>       Debug system bus access supports up to <bits> wide accesses [default 0]
  --dm-auth             Debug module requires debugger to authenticate
  --dmi-rti=<n>         Number of Run-Test/Idle cycles required for a DMI access [default 0]
  --dm-abstract-rti=<n> Number of Run-Test/Idle cycles required for an abstract command to execute [default 0]
  --dm-no-hasel         Debug module supports hasel
  --dm-no-abstract-csr  Debug module won't support abstract to authenticate
  --dm-no-halt-groups   Debug module won't support halt groups
  --dm-no-impebreak     Debug module won't support implicit ebreak in program buffer
  --blocksz=<size>      Cache block size (B) for CMO operations(powers of 2) [default 64]
```

scottj97, 2022-03-18:
You are running `./spike --enable-misaligned`. It's a compile-time option so it needs to be done when configuring: `./configure --enable-misaligned`

scottj97, 2022-03-18:
Perhaps you've already done that -- if so, you're done. There's no runtime option to control misaligned.

aignacio, 2022-03-18:
oh I see, I was thinking it was required to enable it at runtime too, tks for help! =)

closed 2022-03-18 by aignacio