riscv-software-src/riscv-isa-sim #953: Misaligned access

sawansib, 2022-03-23:
Hello!

As per the previous issues and commits RISCV_ENABLE_MISALIGNED is set to false by default.
But if my code has misaligned accesses and I want behavior matching HW that handles misaligned accesses in HW, then 
1) Does Spike support that? If yes how?
2) Or should I just set RISCV_ENABLE_MISALIGNED?

Regards
Sawan

scottj97, 2022-03-23:
When you build Spike, add `--enable-misaligned` to the `./configure` step.

It sounds like we need to add something to the README about the configure options.

sawansib, 2022-03-23:
Thank you for the reply!

What does --enable-misaligned do actually? Does it align the access or create two separate access for the misaligned ones?

Sawan

sawansib, 2022-03-24:
Okay! Thank you for your reply it was really helpful.

Sawan

sawansib, 2022-03-28:
Hi, sorry to re-open this issue. 

As you said that the misaligned load/store is done by a series of one-byte load/stores, do these one-byte loads and stores are real memory operations or some internal Spike workaround for misaligned load/store?

Sawan

scottj97, 2022-03-29:
I'm not sure what you mean by "real" memory operations...as opposed to what?

It's not a workaround; it's a simple implementation of misaligned load/store.

sawansib, 2022-03-30:
I mean if you have a bunch of memory operations doing one byte at a time it will increase the total number of memory operations, right? So this is the way it is implemented in a real RISC machine?

sawansib, 2022-03-30:
Ah okay, thank you for making it clear. 

By "real" I meant Spike is a functional simulator so one load/store should be just one, but when you do the one byte per load/store for a misaligned it will create multiple load/store instead of one.

scottj97, 2022-03-30:
A real hardware design would do two load/store accesses for misaligned. But there is no functional difference between that and one-per-byte like Spike does, unless you're doing misaligned accesses to memory-mapped devices, which no reasonable software should do.

closed 2022-03-30 by sawansib