riscv-software-src/riscv-isa-sim #956: Support for sv57 paging mode

Alenkruth, 2022-03-25:
Hello,

I wanted to know if support for sv57 paging mode is in the works and when will it be available?

Thanks,
Alenkruth

aswaterman, 2022-03-28:
I've made a prototype PR (which I haven't been able to meaningfully test yet).  https://github.com/riscv-software-src/riscv-isa-sim/pull/958

scottj97, 2022-03-29:
@Alenkruth do you have a good way to test #958?

Alenkruth, 2022-03-30:
@scottj97 I do have a way to generate tests with sv57 paging. I will be testing #958 using the tests I am able to generate.

@aswaterman I tried running one test with the sv57 paging. I find the SATP.mode field is not updated with 0xA.
I am adding a snippet from spike's log. Additionally, I am also zipping and attaching the ELF as well as the dump from SPIKE.

> core   0: 3 0x00000000800000e4 (0x01e2b023) mem 0x0000000080008000 0x0000000020002401
> core   0: 0x00000000800000e8 (0x00a00293) li      t0, 10
> core   0: 3 0x00000000800000e8 (0x00a00293) x 5 0x000000000000000a
> core   0: 0x00000000800000ec (0x03c29313) slli    t1, t0, 60
> core   0: 3 0x00000000800000ec (0x03c29313) x 6 0xa000000000000000
> core   0: 0x00000000800000f0 (0x18032073) csrs    satp, t1
> core   0: 3 0x00000000800000f0 (0x18032073) c384_satp 0x0000000000000000
> core   0: 0x00000000800000f4 (0x00003e17) auipc   t3, 0x3
> core   0: 3 0x00000000800000f4 (0x00003e17) x28 0x00000000800030f4
> core   0: 0x00000000800000f8 (0xf0ce0e13) addi    t3, t3, -244
> core   0: 3 0x00000000800000f8 (0xf0ce0e13) x28 0x0000000080003000
> core   0: 0x00000000800000fc (0x00ce5e93) srli    t4, t3, 12
> core   0: 3 0x00000000800000fc (0x00ce5e93) x29 0x0000000000080003
> core   0: 0x0000000080000100 (0x01d00f33) add     t5, zero, t4
> core   0: 3 0x0000000080000100 (0x01d00f33) x30 0x0000000000080003
> core   0: 0x0000000080000104 (0x180f2073) csrs    satp, t5
> core   0: 3 0x0000000080000104 (0x180f2073) c384_satp 0x0000000000080003
> core   0: 0x0000000080000108 (0x00100f93) li      t6, 1
> core   0: 3 0x0000000080000108 (0x00100f93) x31 0x0000000000000001

[sv57.tar.gz](https://github.com/riscv-software-src/riscv-isa-sim/files/8377517/sv57.tar.gz)

aswaterman, 2022-03-30:
The default maximum is still Sv48.  Currently, the only way to enable Sv57 is to pass in a devicetree blob (`--dtb`) with the `mmu-type` overridden to `riscv,sv57`.

Alenkruth, 2022-03-30:
Passing a device tree blob with the changes you suggested seems to work!!

Spike was able to execute the test with sv57 paging. I am attaching the ELF and log as a zip.

I will be running some more tests on spike. Are there any specific tests you want me to run?

[sv57_working.tar.gz](https://github.com/riscv-software-src/riscv-isa-sim/files/8378419/sv57_working.tar.gz)

aswaterman, 2022-03-30:
Great!  No, I can't recommend any additional test vectors.  I'll wait for you to report back about your additional testing before I merge the PR.

Alenkruth, 2022-03-30:
I was able to run another 15 tests with sv57 enabled. I did not face any issues as of now.

I think you can merge the PR. I will continue my testing and I will create an issue if I find any odd behavior.

Thanks,
Alenkruth

aswaterman, 2022-03-30:
OK, thanks.  I'll close this, too.

closed 2022-03-30 by aswaterman