riscv-software-src/riscv-isa-sim #961: Check in PATH when running target program

arunisaac, 2022-04-01:
When running a target program, spike should look for that program in paths specified by the `PATH` environment variable as well.

Currently, when I run `spike pk` with both `spike` and `pk` being in `PATH`, I get the following error. When I run the same command with the absolute path to `pk`, it works fine.

```
$ spike pk ...
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not open pk (did you misspell it? If VCS, did you forget +permissive/+permissive-off?)
```

Thanks!

aswaterman, 2022-04-01:
I don't think this is a good idea.  Your PATH should contain binaries you're interested in executing on your host machine, but pk is a binary you should only ever use in conjunction with a simulator.

arunisaac, 2022-04-01:
My question is in the context of packaging spike and pk for Guix, a
GNU/Linux distribution. Guix installs executables to non-standard paths
under /gnu/store such as
/gnu/store/sn650n6mzgz4f0sqc9b18d0gp4mls5ld-riscv-pk-1.0.0/bin/pk . To
run spike pk, I have to run something like

$ spike
/gnu/store/sn650n6mzgz4f0sqc9b18d0gp4mls5ld-riscv-pk-1.0.0/bin/pk ...

Note that I don't need to specify the absolute path to spike since it is
in PATH.

> I don't think this is a good idea.  Your PATH should contain binaries
> you're interested in executing on your host machine, but pk is a
> binary you should only ever use in conjunction with a simulator.

Ok, that's reasonable.

But, would a different environment variable, say SPIKE_PATH,
SPIKE_CROSS_PATH or similar, be acceptable?

Also, is pk meant to be used exclusively with spike, or does it have a
larger function elsewhere?

aswaterman, 2022-04-01:
That is indeed an awkwardly long path.  But perhaps this is a problem better solved by writing a shell script or an alias, rather than modifying Spike.

> Also, is pk meant to be used exclusively with spike, or does it have a larger function elsewhere?

Among other things, it works with rocket-chip (https://github.com/chipsalliance/rocket-chip) and various FPGA and ASIC designs.

arunisaac, 2022-04-01:
Guix lets you do something like this:

$ spike $(guix build riscv-pk)/bin/pk ...

One can live with that. I was just hoping we could improve the situation
a little bit. But, not to worry too much.

closed 2022-04-01 by arunisaac