riscv-software-src/riscv-isa-sim #967: Runtime error when simulating SPEC2017 (bad syscall #179)

GuangdaLiu, 2022-04-08:
I'm trying to simulate some benchmarks from [SPEC2017](https://www.spec.org/cpu2017/Docs/overview.html#Q13).

Some simpler benchmarks can be simulated successfully, but when simulating some more complex benchmarks (e.g. 502.gcc_r), they crashed quickly. All of those errors are **`bad syscall #179!`**. I have no idea about what does that mean. 

Those benchmarks are built with `riscv64-unknown-linux-gnu-gcc (--with-abi=lp64) -static -O3`. And I also tested building with `-O0`, but the runtime error also occured.

aswaterman, 2022-04-08:
In general, PK doesn't support the full set of Linux system calls; it only supports enough to get basic programs running.  It looks like this system call is `sysinfo`.

I've pushed a patch that stubs out `sysinfo` (see d6008c14176bd663e7def6b89679b160af4815c2).  No guarantees it'll work, but you should give it a shot.

If you need to stub out additional syscalls, you can follow the same pattern and make a pull request.

aswaterman, 2022-04-08:
BTW, this is the wrong issue tracker; the error you see is in `riscv-pk`, not Spike.

GuangdaLiu, 2022-04-10:
@aswaterman Thanks a lot for your effort and advice.

But it doesn't seems to work for me. Even though I rebuilt and installed both `pk` and `spike`, I still get that error :(

GuangdaLiu, 2022-04-11:
> @aswaterman Thanks a lot for your effort and advice.
> 
> But it doesn't seems to work for me. Even though I rebuilt and installed both `pk` and `spike`, I still get that error :(
> 
> Actually although I modify `panic("bad syscall #%ld!",n);` in `syscall.c` directly, the error does not change.
> 
> So I guess the change of `pk` is not applied to `spike`, even though both are rebuilt and installed?

Oh, `spike` seems to look for `pk` in `$RISCV/riscv64-unknown-elf/bin/`, even though I built `pk` with `riscv64-unknown-linux-gnu` and installed it in `$RISCV/riscv64-unknown-linux-gnu/bin/`. 

When I provide the absolute path of latest `pk`, the change is applied and the error does not occur again.

closed 2022-04-11 by GuangdaLiu

aswaterman, 2022-04-11:
Glad it worked.  Thanks for the bug report!