riscv-software-src/riscv-isa-sim #970: ePMP support

rswarbrick, 2022-04-11:
Are there any imminent plans to add [ePMP](https://github.com/riscv/riscv-tee/blob/main/Smepmp/Smepmp.pdf) support to Spike? At lowRISC, we have a processor that supports it and would like to extend Spike to be able to verify things properly.

I know that there is a set of patches for ePMP support at https://github.com/joxie/riscv-isa-sim, but a quick look at the patches suggests these aren't in an upstreamable form.

Is anyone else working on this at the moment, or should we roll up our proverbial sleeves?

aswaterman, 2022-04-11:
FWIW, I hadn't planned to work on this any time soon, so lowRISC contributing a patch set would be welcome.

rswarbrick, 2022-04-12:
Great, thanks for getting back to us. We'll start hacking soon.

joxie, 2022-04-18:
Hi Rupert I have a working fork at https://github.com/joxie/riscv-isa-sim and plan to upstream soon

aswaterman, 2022-04-18:
@rswarbrick since you have a good idea of our process and expectations, can I ask that you work with @joxie to shepherd this PR?

joxie, 2022-04-18:
https://github.com/joxie/riscv-isa-sim/tree/epmp is cleaner (without tests, need to find a home for tests)

joxie, 2022-04-19:
@rswarbrick please let me know the flow to follow for this PR?

rswarbrick, 2022-04-20:
Well, the first step is probably to fiddle around to get a series of patches rebased cleanly on to the upstream master branch. Once we've got that, we can take a look at the commits that need upstreaming and try to figure out how to chop them up into small PRs that can be merged incrementally. (If you try to do the whole lot at once, it's almost impossible to avoid getting bogged down in interminable reviews in any project that has proper PR reviewing)

joxie, 2022-04-24:
I created PR #980

joxie, 2022-05-05:
PR upstreamed, we can close this issue now

scottj97, 2022-05-05:
Implemented in #983

rswarbrick, 2022-05-06:
Late to the party but... that's absolutely awesome! Thank you so much for doing this.