riscv-software-src/riscv-isa-sim #982: Inconsistent reporting of csr values in log.

pawks, 2022-04-27:
For certain instructions, the `fflags` value is not printed if the new value is equal to the value being written. But in other cases the `fflags` value is printed in the output irrespective of whether the values are the same.
 
Snippet 1(new value is printed only if it is different from old value):
Instructions at `1c4` and `1f2` are both `fsub.d` instructions. However the `1c4` instruction does not set any exception flags.

```
core   0: 0 0x000000008000019e (0x001015f3) x11 0x0000000000000001 c1_fflags 0x0000000000000000
core   0: 0 0x00000000800001a2 (0x4605) x12 0x0000000000000001
core   0: 0 0x00000000800001a4 (0x14d51763)
core   0: 0 0x00000000800001a8 (0x14c59563)
core   0: 0 0x00000000800001ac (0x4195) x 3 0x0000000000000005
core   0: 0 0x00000000800001ae (0x00002517) x10 0x00000000800021ae
core   0: 0 0x00000000800001b2 (0xeb250513) x10 0x0000000080002060
core   0: 0 0x00000000800001b6 (0x00053007) f 0 0x4004000000000000 mem 0x0000000080002060
core   0: 0 0x00000000800001ba (0x00853087) f 1 0x3ff0000000000000 mem 0x0000000080002068
core   0: 0 0x00000000800001be (0x01053107) f 2 0x0000000000000000 mem 0x0000000080002070
core   0: 0 0x00000000800001c2 (0x6d14) x13 0x3ff8000000000000 mem 0x0000000080002078
core   0: 0 0x00000000800001c4 (0x0a1071d3) f 3 0x3ff8000000000000
core   0: 0 0x00000000800001c8 (0xe2018553) x10 0x3ff8000000000000
core   0: 0 0x00000000800001cc (0x001015f3) x11 0x0000000000000000 c1_fflags 0x0000000000000000
core   0: 0 0x00000000800001d0 (0x4601) x12 0x0000000000000000
core   0: 0 0x00000000800001d2 (0x12d51063)
core   0: 0 0x00000000800001d6 (0x10c59e63)
core   0: 0 0x00000000800001da (0x4199) x 3 0x0000000000000006
core   0: 0 0x00000000800001dc (0x00002517) x10 0x00000000800021dc
core   0: 0 0x00000000800001e0 (0xea450513) x10 0x0000000080002080
core   0: 0 0x00000000800001e4 (0x00053007) f 0 0xc0934c6666666666 mem 0x0000000080002080
core   0: 0 0x00000000800001e8 (0x00853087) f 1 0xbff199999999999a mem 0x0000000080002088
core   0: 0 0x00000000800001ec (0x01053107) f 2 0x0000000000000000 mem 0x0000000080002090
core   0: 0 0x00000000800001f0 (0x6d14) x13 0xc093480000000000 mem 0x0000000080002098
core   0: 0 0x00000000800001f2 (0x0a1071d3) c1_fflags 0x0000000000000001 f 3 0xc093480000000000
```

Snippet 2(value printed unconditionally):
Instructions at `7c6` and `7e0` are both `fcvt.wu.d`, the `fflags` value is `0x10` before `7c6` too. The `fflags` seem to be printed unconditionally for all instructions which update the integer register file. 

```
core   0: 0 0x00000000800007c6 (0xc210f0d3) c1_fflags 0x0000000000000010 x 1 0xffffffffffffffff
core   0: 0 0x00000000800007ca (0x53fd) x 7 0xffffffffffffffff
core   0: 0 0x00000000800007cc (0x03e00193) x 3 0x000000000000003e
core   0: 0 0x00000000800007d0 (0x0a709f63)
core   0: 0 0x00000000800007d4 (0x00002097) x 1 0x00000000800027d4
core   0: 0 0x00000000800007d8 (0xc7c08093) x 1 0x0000000080002450
core   0: 0 0x00000000800007dc (0x0080b087) f 1 0x7fffffffffffffff mem 0x0000000080002458
core   0: 0 0x00000000800007e0 (0xc210f0d3) c1_fflags 0x0000000000000010 x 1 0xffffffffffffffff
```

Similarly, one of the older spike versions seemed to output the `mstatus` values for all floating point instructions. But the latest version seems to ommit it(I assume it is because the new value and the updated value are the same). Although the issue of `fflags` not being printed persists here too. 
Disassembly:
```
    800001b6:	00053007          	fld	ft0,0(a0)
    800001ba:	00853087          	fld	ft1,8(a0)
    800001be:	01053107          	fld	ft2,16(a0)
    800001c2:	6d14                	ld	a3,24(a0)
    800001c4:	0a1071d3          	fsub.d	ft3,ft0,ft1
```
Latest:
```
core   0: 0 0x00000000800001b6 (0x00053007) f 0 0x4004000000000000 mem 0x0000000080002060
core   0: 0 0x00000000800001ba (0x00853087) f 1 0x3ff0000000000000 mem 0x0000000080002068
core   0: 0 0x00000000800001be (0x01053107) f 2 0x0000000000000000 mem 0x0000000080002070
core   0: 0 0x00000000800001c2 (0x6d14) x13 0x3ff8000000000000 mem 0x0000000080002078
core   0: 0 0x00000000800001c4 (0x0a1071d3) f 3 0x3ff8000000000000
```
Older:
```
core   0: 0 0x00000000800001b6 (0x00053007) c768_mstatus 0x8000000a00006080 f 0 0x4004000000000000 mem 0x0000000080002060
core   0: 0 0x00000000800001ba (0x00853087) c768_mstatus 0x8000000a00006080 f 1 0x3ff0000000000000 mem 0x0000000080002068
core   0: 0 0x00000000800001be (0x01053107) c768_mstatus 0x8000000a00006080 f 2 0x0000000000000000 mem 0x0000000080002070
core   0: 0 0x00000000800001c2 (0x6d14) x13 0x3ff8000000000000 mem 0x0000000080002078
core   0: 0 0x00000000800001c4 (0x0a1071d3) c768_mstatus 0x8000000a00006080 f 3 0x3ff8000000000000
```

Is there any reason for these inconsistencies? IMO, all csr updates which occur as a side effect of an instruction should be printed in the commit log, irrespective of whether the value being written is different than the one already present. Many RTLs use the commit-logs from spike as a smoke test for verifying functionality. It would be helpful if the scheme for printing csr updates is uniform and unconditional.

scottj97, 2022-04-27:
For both `fsub.d` and `fcvt.wu.d`, the fflags are updated via this macro in `decode.h`:
```
#define set_fp_exceptions ({ if (softfloat_exceptionFlags) { \
                               STATE.fflags->write(STATE.fflags->read() | softfloat_exceptionFlags); \
                             } \
                             softfloat_exceptionFlags = 0; })
```

So if no flags were set by the current instruction, no write will be logged. You can remove the `if()` to always write fflags, but it might significantly impact performance. (See #950)

I guess in your code, the `fcvt.wu.d` is setting flags that are already set, so no change occurs even though it logs a write. Whereas the `fsub.d` is not trying to set any flags.

For mstatus, how old is "older Spike"? I [added mstatus.FS logging](https://github.com/riscv-software-src/riscv-isa-sim/commit/35140dff22ab4487d8a47f016253142bb8e5666f) as part of #793...prior to that those writes were not properly logged.

For better performance, we [removed redundant mstatus.FS dirtying](https://github.com/riscv-software-src/riscv-isa-sim/commit/d266b96018d5edb6f8f072e14c66a6595ed68491) as part of #950.

For testing RTL, you will no doubt need to make many changes to Spike to adapt it to your platform choices. When exactly these unnecessary/redundant CSR writes are reported will be peculiar to your design and verification environment. (I use Spike this way, and I've made several such changes in my fork of Spike.)

I don't consider it a Spike bug unless a CSR changes with no record in the commit log.

pawks, 2022-04-27:
> So if no flags were set by the current instruction, no write will be logged. You can remove the `if()` to always write fflags, but it might significantly impact performance. (See #950)
This makes sense now. Thanks! 

> For mstatus, how old is "older Spike"? I [added mstatus.FS logging](https://github.com/riscv-software-src/riscv-isa-sim/commit/35140dff22ab4487d8a47f016253142bb8e5666f) as part of #793...prior to that those writes were not properly logged.
> 
I only have the spike binary. I do not have the accurate commit number or the date at which the binary was generated.  
> For testing RTL, you will no doubt need to make many changes to Spike to adapt it to your platform choices. When exactly these unnecessary/redundant CSR writes are reported will be peculiar to your [...]
Does it make sense to enable/disable these features via a macro/hook in the configure script. The default behavior remains geared towards performance, while the hook enables reporting csr writes unconditionally? I can raise a PR if such a feature is usefull.

scottj97, 2022-04-27:
> Does it make sense to enable/disable these features via a macro/hook in the configure script. The default behavior remains geared towards performance, while the hook enables reporting csr writes unconditionally? I can raise a PR if such a feature is usefull.

I'd rather not support another configure flag. Each one doubles the number of Spike builds required for thorough testing.

My point was that since you'll need to maintain your own local fork of Spike anyway, whatever CSR logging behavior you need can easily be made (and maintained) in that fork.

closed 2022-04-27 by pawks

omerguzelelectronicguy, 2023-07-21:
> For both `fsub.d` and `fcvt.wu.d`, the fflags are updated via this macro in `decode.h`:
>
> ```
> #define set_fp_exceptions ({ if (softfloat_exceptionFlags) { \ [...]

Hello @scottj97 . I have a question on that. I don't want to lose the performance instead I wanted to make my rtl same with the spike FPU csr log.  However, I couldn't understand which cases the FFLAGS are logged. Should I understand by looking the code arround you mentioned above? or is there explaination about the log systematic.

scottj97, 2023-07-24:
> Hello @scottj97 . I have a question on that. I don't want to lose the performance instead I wanted to make my rtl same with the spike FPU csr log. However, I couldn't understand which cases the [...]

Based on my earlier comment, it appears the fflags are logged when any instruction tries to set any bit, even if that bit is already set.