riscv-software-src/riscv-isa-sim #984: Why spike -l output is different from objdump result?

hidetatz, 2022-04-30:
I am trying to run [riscv-tests](https://github.com/riscv-software-src/riscv-tests) test program using spike.
However, I can see that the execution log of spike is different from its objdump result.
I manually built the riscv-tests. For example, when I try to run "rv64ui-p-add" ELF binary, "spike -l" shows the below logs:

```
core   0: 0x0000000000001000 (0x00000297) auipc   t0, 0x0
core   0: 0x0000000000001004 (0x02028593) addi    a1, t0, 32
core   0: 0x0000000000001008 (0xf1402573) csrr    a0, mhartid
core   0: 0x000000000000100c (0x0182b283) ld      t0, 24(t0)
core   0: 0x0000000000001010 (0x00028067) jr      t0
core   0: 0x0000000080000000 (0x1f80006f) j       pc + 0x1f8
core   0: 0x00000000800001f8 (0x00000093) li      ra, 0
core   0: 0x00000000800001fc (0x00000113) li      sp, 0
core   0: 0x0000000080000200 (0x00000193) li      gp, 0
core   0: 0x0000000080000204 (0x00000213) li      tp, 0
core   0: 0x0000000080000208 (0x00000293) li      t0, 0
core   0: 0x000000008000020c (0x00000313) li      t1, 0
core   0: 0x0000000080000210 (0x00000393) li      t2, 0
core   0: 0x0000000080000214 (0x00000413) li      s0, 0
core   0: 0x0000000080000218 (0x00000493) li      s1, 0
core   0: 0x000000008000021c (0x00000613) li      a2, 0
core   0: 0x0000000080000220 (0x00000693) li      a3, 0
core   0: 0x0000000080000224 (0x00000713) li      a4, 0
(trimmed because it's long)
```

However, the objdump result of "rv64ui-p-add" (`riscv64-unknown-elf-objdump -d ./tests/riscv-tests/rv64ui-p-add`) is:

```

./tests/riscv-tests/rv64ui-p-add:     file format elf64-littleriscv

Disassembly of section .text.init:

0000000080000000 <_start>:
    80000000:   04c0006f                j       8000004c <reset_vector>

0000000080000004 <trap_vector>:
    80000004:   34202f73                csrr    t5,mcause
    80000008:   00800f93                li      t6,8
    8000000c:   03ff0a63                beq     t5,t6,80000040 <write_tohost>
    80000010:   00900f93                li      t6,9
    80000014:   03ff0663                beq     t5,t6,80000040 <write_tohost>
    80000018:   00b00f93                li      t6,11
    8000001c:   03ff0263                beq     t5,t6,80000040 <write_tohost>
    80000020:   80000f17                auipc   t5,0x80000
    80000024:   fe0f0f13                addi    t5,t5,-32 # 0 <_start-0x80000000>
    80000028:   000f0463                beqz    t5,80000030 <trap_vector+0x2c>
    8000002c:   000f0067                jr      t5
    80000030:   34202f73                csrr    t5,mcause
    80000034:   000f5463                bgez    t5,8000003c <handle_exception>
    80000038:   0040006f                j       8000003c <handle_exception>

000000008000003c <handle_exception>:
    8000003c:   5391e193                ori     gp,gp,1337

0000000080000040 <write_tohost>:
    80000040:   00001f17                auipc   t5,0x1
    80000044:   fc3f2023                sw      gp,-64(t5) # 80001000 <tohost>
    80000048:   ff9ff06f                j       80000040 <write_tohost>

000000008000004c <reset_vector>:
    8000004c:   f1402573                csrr    a0,mhartid
    80000050:   00051063                bnez    a0,80000050 <reset_vector+0x4>
    80000054:   00000297                auipc   t0,0x0
    80000058:   01028293                addi    t0,t0,16 # 80000064 <reset_vector+0x18>
    8000005c:   30529073                csrw    mtvec,t0
    80000060:   18005073                csrwi   satp,0
    80000064:   00000297                auipc   t0,0x0
    80000068:   02428293                addi    t0,t0,36 # 80000088 <reset_vector+0x3c>
    8000006c:   30529073                csrw    mtvec,t0
(trimmed)
```

It seems like the program counter and the executed instruction are both different.
I guess I am missing something, but can you please tell me why this is happening?

scottj97, 2022-04-30:
Spike has a small boot ROM at 0x1000; that's the first instructions you see in the commit log.

At 0x80000000 the program begins. Spike sees an opcode of 0x1f80006f while the ELF says it should be 0x04c0006f. I can't explain that.

The sequence of code in the commit log looks reasonable. It's running the intended program, not garbage.

My guess is you're disassembling a different binary than the one given to Spike.

hidetatz, 2022-05-01:
Thank you. Now I understand that there are some instructions for boot.
I tried building riscv-tests again just in case and checked the result but it still shows the different output (although it's objdump result is a bit changed). This is the reproduce step which I tried:

```
# build riscv-tests
git clone https://github.com/riscv/riscv-tests
cd riscv-tests
git submodule update --init --recursive
autoconf
./configure --prefix=/opt/riscv/target
make
make install

# This is riscv-tests binary objdump result
riscv64-unknown-elf-objdump -d /opt/riscv/target/share/riscv-tests/isa/rv64ui-p-add > rv64ui-p-add-dump

# This is spike log of the same binary execution log by spike
spike -l pk /opt/riscv/target/share/riscv-tests/isa/rv64ui-p-add 2> rv64ui-p-add-spike-log
```

aswaterman, 2022-05-01:
You are seeing the execution log of the program ‘pk’, not the test itself. These tests are meant to be run bare-metal, not atop pk.

hidetatz, 2022-05-01:
Ah, I found that running spike without pk (`spike -l /opt/riscv/target/share/riscv-tests/isa/rv64ui-p-add`) shows the same instruction logs with objdump result.
Thank you for your help @scottj97 @aswaterman . I can close the issue now.

closed 2022-05-01 by hidetatz