riscv-software-src/riscv-isa-sim #1002: Question about lr and sc instruction

YangGuangRongXian, 2022-05-16:
In riscv unprivileged spec 9.2, below 4 rules are mentioned:
1.  The SC must fail if the address is not within the reservation set of the most recent LR in program order.
2.  An implementation can register an arbitrarily large reservation set on each LR, provided the reservation set includes all bytes of the 
     addressed data word or doubleword.
3.  LR might have had a different effective address and data size, but reserved the SC's address as part of the reservation set.
4.  The SC succeeds only if the reservation is still valid and the reservation set contains the bytes being written.

For RV64,  assume now we have 2 different implementations:
1. LR.W  acquire 8 bytes  reservation set and LR.D   acquire 8 bytes  reservation set, this is legal according to rule 2.
2. LR.W  acquire 4 bytes  reservation set and LR.D   acquire 8 bytes  reservation set.

And assume lr && sc access the same aligned address, is below conclusion right?
In implement 1, if  execute lr.w && sc.d, then sc.d will success. If  execute lr.d && sc.w, sc.w will success too.
In implement 2, if  execute lr.w && sc.d, then sc.d will fail because not satisfy rule 4. If  execute lr.d & sc.w, sc.w will success.

aswaterman, 2022-05-16:
This seems like a question that's more appropriate for [the mailing lists like https://groups.google.com/a/groups.riscv.org/g/isa-dev -- it is not Spike-specific.  Please ask any follow-up questions there.  But, for the record, both implementations 1 and 2 sound legal to me.

closed 2022-05-16 by aswaterman

YangGuangRongXian, 2022-05-16:
Ok, thanks a lot, and one more doubt about the statement in:
https://github.com/riscv/riscv-isa-manual/blob/a2cf7ea4cbe84f7ec25081cddf7c5462fb0b737a/src/a.tex#164-L166

For a specific implement, it's  reservation set size that lr register is fixed, even though it can choose an 'arbitrarily large' size, but the size will be certain once it is choosed, is this understanding right?