riscv-software-src/riscv-isa-sim #1005: Memory address is invalid

athurwo, 2022-05-18:
I want to simulate elf file like below:
`spike --isa=rv32imc --priv=m /opt/riscv-elf-32/riscv32-unknown-elf/bin/pk ./vcs_results/default/corev_rand_instr_test/0/test_program/corev_rand_instr_test_0.elf`
result is :
Access exception occurred while host was accessing memory on behalf of target (tohost = 0xd8000c060):
Memory address 0xd8000c060 is invalid

`spike --isa=rv32imc --priv=m ./vcs_results/default/corev_rand_instr_test/0/test_program/corev_rand_instr_test_0.elf`
result is:
Access exception occurred while loading payload ./vcs_results/default/corev_rand_instr_test/0/test_program/corev_rand_instr_test_0.elf:
Memory address 0x80 is invalid

Who can help me resolve this problem?
Thanks.

scottj97, 2022-05-18:
Spike has a limited memory range. You need to tell it using `-m` where you want memories. If you really want to use 0x80 you will need to put a memory there.

But the first problem is a little different because apparently the `tohost` label is outside of the 34-bit physical address range of an RV32 machine. You will need to analyze that ELF file using tools from binutils (such as `riscv32-unknown-elf-objdump`) and understand how it was compiled in order to fix that.

scottj97, 2022-05-19:
This might be due to interference with Spike's debug ROM? I'm afraid I don't know much about that.

aswaterman, 2022-05-20:
Yes, that sounds right. Not a bug, but really a platform constraint on
where Spike allows memory to be located.

athurwo, 2022-05-20:
> This might be due to interference with Spike's debug ROM? I'm afraid I don't know much about that.

yes, 0x0~0x800 is debug which have 4k,  I have change to main that is 0x80000000~0xFFFFFFFF:
`spike --isa=rv32imc --priv=m -m0x80000000:0x100000000 /opt/riscv-elf-32/riscv32-unknown-elf/bin/pk ./vcs_results/default/corev_rand_instr_test/0/test_program/corev_rand_instr_test_0.elf`
the error is changed:
Access exception occurred while host was accessing memory on behalf of target (tohost = 0xd8000c060):
Memory address 0xd8000c060 is invalid

athurwo, 2022-05-20:
> This might be due to interference with Spike's debug ROM? I'm afraid I don't know much about that.

Do you have some data or infomation or files about spike memory ?
I want to learn it, then try to solve this problem.

scottj97, 2022-05-20:
> Memory address 0xd8000c060 is invalid

This is the first problem I mentioned above in my first comment. Have you attempted to figure out where this too-large address is coming from? The problem seems to be with the compilation or linking of `corev_rand_instr_test_0.elf`.

I don't know of any documentation on Spike memories. There is the debug ROM at 0, a boot ROM at 0x1000, and the CLINT at 0x02000000-0x020c0000. The default RAM is at 0x80000000.

athurwo, 2022-05-21:
> This is the first problem I mentioned above in my first comment. Have you attempted to figure out where this too-large address is coming from? The problem seems to be with the compilation or linking [...]

thanks, I should check the elf generation carefully.

athurwo, 2022-05-21:
> This is the first problem I mentioned above in my first comment. Have you attempted to figure out where this too-large address is coming from? The problem seems to be with the compilation or linking [...]

reinstall spike and pk with isa=rv32imc and priv=m, now execute command:
`spike --isa=rv32imc --priv=m /opt/riscv-elf-32/riscv32-unknown-elf/bin/pk ./vcs_results/default/corev_rand_instr_test/0/test_program/corev_rand_instr_test_0.elf`
result is :
bbl loader
../machine/mtrap.c:22: machine mode: unhandlable trap 2 @ 0x80001acc
Power off

now it looks ok that has bbl loader and power off， I am going on to fix mcause=2, and mepc=0x80001acc through objdump file.

closed 2022-05-24 by scottj97