riscv-software-src/riscv-isa-sim #1007: Something is error when i make in /riscv-isa-sim/build

hust-gaoyujing, 2022-05-19:
When i run `./script/build-spike.sh` in Gemmini, something is wrong.So i run the command line one by one,and i find that the error happened when `make` in `toolchains/esp-tools/riscv-isa-sim/build`.

./script/build-spike.sh:
```
export GEMMINI_ONLY_GENERATE_GEMMINI_H=1

cd ../../sims/verilator/
echo Generating new gemmini_params.h file...
make verilog CONFIG=CustomGemminiSoCConfig &> build.log

cd -
cp software/gemmini-rocc-tests/include/gemmini_params.h ../../toolchains/esp-tools/riscv-isa-sim/gemmini/gemmini_params.h
cd ../../toolchains/esp-tools/riscv-isa-sim/build
make && make install
```

error massage:
```
Makefile:349: warning: overriding recipe for target 'disasm.o'
Makefile:349: warning: ignoring old recipe for target 'disasm.o'
make: Circular libhwacha.so <- libhwacha.so dependency dropped.
g++ -MMD -MP -I/include/boost-0 -DPREFIX=\"/home/gaoyujing/chipyard/esp-tools-install\" -Wall -Wno-unused -g -O2 -fPIC -std=c++11 -g -O2   -I. -I.. -I../fesvr -I../riscv -I../hwacha -I../gemmini -I../sha3 -I../disasm -I../customext -I../fdt -I../softfloat -I../spike_main -I../spike_dasm  -fPIC -c ../gemmini/gemmini.cc
../gemmini/gemmini.cc:420:163: error: macro "assert" passed 2 arguments, but takes just 1
  420 | rows[state_id] <= 1, "If Gemmini is built without first-layer optimizations, then 'pixels_per_rows' cannot be larger than 1\n");
      |                                                                                                                               ^

In file included from ../gemmini/gemmini.cc:6:
/usr/include/assert.h:92: note: macro "assert" defined here
   92 | #  define assert(expr)       \
      | 
../gemmini/gemmini.cc:906:143: error: macro "assert" passed 2 arguments, but takes just 1
  906 | els_per_row <= 1, "If Gemmini is built without first-layer optimizations, then 'max_pixels_per_row' cannot be larger than 1\n");
      |                                                                                                                               ^

In file included from ../gemmini/gemmini.cc:6:
/usr/include/assert.h:92: note: macro "assert" defined here
   92 | #  define assert(expr)       \
      | 
../gemmini/gemmini.cc: In member function ‘void gemmini_t::config(reg_t, reg_t)’:
../gemmini/gemmini.cc:420:5: error: ‘assert’ was not declared in this scope
  420 |     assert(gemmini_state.pixels_per_rows[state_id] <= 1, "If Gemmini is built without first-layer optimizations, then 'pixels_per_rows' cannot be larger than 1\n");
      |     ^~~~~~
../gemmini/gemmini.cc:7:1: note: ‘assert’ is defined in header ‘<cassert>’; did you forget to ‘#include <cassert>’?
    6 | #include <assert.h>
  +++ |+#include <cassert>
    7 | 
../gemmini/gemmini.cc: In member function ‘void gemmini_t::loop_conv_ws(reg_t, reg_t)’:
../gemmini/gemmini.cc:906:3: error: ‘assert’ was not declared in this scope
  906 |   assert(max_pixels_per_row <= 1, "If Gemmini is built without first-layer optimizations, then 'max_pixels_per_row' cannot be larger than 1\n");
      |   ^~~~~~
../gemmini/gemmini.cc:906:3: note: ‘assert’ is defined in header ‘<cassert>’; did you forget to ‘#include <cassert>’?
make: *** [Makefile:349: gemmini.o] Error 1
gaoyujing@LAPTOP-MA3NN146:~/chipyard/toolchains/esp-tools/riscv-isa-sim/build$ make 
Makefile:349: warning: overriding recipe for target 'disasm.o'
Makefile:349: warning: ignoring old recipe for target 'disasm.o'
make: Circular libhwacha.so <- libhwacha.so dependency dropped.
g++ -MMD -MP -I/include/boost-0 -DPREFIX=\"/home/gaoyujing/chipyard/esp-tools-install\" -Wall -Wno-unused -g -O2 -fPIC -std=c++11 -g -O2   -I. -I.. -I../fesvr -I../riscv -I../hwacha -I../gemmini -I../sha3 -I../disasm -I../customext -I../fdt -I../softfloat -I../spike_main -I../spike_dasm  -fPIC -c ../gemmini/gemmini.cc
../gemmini/gemmini.cc:420:163: error: macro "assert" passed 2 arguments, but takes just 1
  420 | rows[state_id] <= 1, "If Gemmini is built without first-layer optimizations, then 'pixels_per_rows' cannot be larger than 1\n");
      |                                                                                                                               ^

In file included from /usr/include/c++/9/cassert:44,
                 from ../gemmini/gemmini.cc:7:
/usr/include/assert.h:92: note: macro "assert" defined here
   92 | #  define assert(expr)       \
      | 
../gemmini/gemmini.cc:906:143: error: macro "assert" passed 2 arguments, but takes just 1
  906 | els_per_row <= 1, "If Gemmini is built without first-layer optimizations, then 'max_pixels_per_row' cannot be larger than 1\n");
      |                                                                                                                               ^

In file included from /usr/include/c++/9/cassert:44,
                 from ../gemmini/gemmini.cc:7:
/usr/include/assert.h:92: note: macro "assert" defined here
   92 | #  define assert(expr)       \
      | 
../gemmini/gemmini.cc: In member function ‘void gemmini_t::config(reg_t, reg_t)’:
../gemmini/gemmini.cc:420:5: error: ‘assert’ was not declared in this scope
  420 |     assert(gemmini_state.pixels_per_rows[state_id] <= 1, "If Gemmini is built without first-layer optimizations, then 'pixels_per_rows' cannot be larger than 1\n");
      |     ^~~~~~
../gemmini/gemmini.cc:8:1: note: ‘assert’ is defined in header ‘<cassert>’; did you forget to ‘#include <cassert>’?
    7 | #include <cassert>
  +++ |+#include <cassert>
    8 | using namespace std;
../gemmini/gemmini.cc: In member function ‘void gemmini_t::loop_conv_ws(reg_t, reg_t)’:
../gemmini/gemmini.cc:906:3: error: ‘assert’ was not declared in this scope
  906 |   assert(max_pixels_per_row <= 1, "If Gemmini is built without first-layer optimizations, then 'max_pixels_per_row' cannot be larger than 1\n");
      |   ^~~~~~
../gemmini/gemmini.cc:906:3: note: ‘assert’ is defined in header ‘<cassert>’; did you forget to ‘#include <cassert>’?
make: *** [Makefile:349: gemmini.o] Error 1
```

The error massage said `note: ‘assert’ is defined in header ‘<cassert>’; did you forget to ‘#include <cassert>’?` ,but it will also remind me to add <cassert> when i added `#include <cassert>`.

scottj97, 2022-05-19:
The error is coming from `gemmini.cc` which is not part of this repo.

hust-gaoyujing, 2022-05-19:
I think that there are some syntax errors in `gemmini.cc`(come from the gemmini)，and i modify it.Now it works.

closed 2022-05-19 by scottj97