riscv-software-src/riscv-isa-sim #1008: Writing specific address of memory causes trap_entry

omerguzelelectronicguy, 2022-05-20:
Hello,
I am beginner spike user in bare metal mode for RISC-V. I wanted to write data to the specific address of memory which is 0x20000000 in Spike. When I used the command below:
`spike --log-commits -l -d  hword.riscv` it gives an output below. (I only added after main to trap_entry.)
```
core   0: >>>>  main
core   0: 0x00000000800029d4 (0xff010113) addi    sp, sp, -16
core   0: 3 0x00000000800029d4 (0xff010113) x2  0x0000000080022b40
: 
core   0: 0x00000000800029d8 (0x00113423) sd      ra, 8(sp)
core   0: 3 0x00000000800029d8 (0x00113423) mem 0x0000000080022b48 0x0000000080002818
: 
core   0: 0x00000000800029dc (0x200007b7) lui     a5, 0x20000
core   0: 3 0x00000000800029dc (0x200007b7) x15 0x0000000020000000
: 
core   0: 0x00000000800029e0 (0x00700713) li      a4, 7
core   0: 3 0x00000000800029e0 (0x00700713) x14 0x0000000000000007
: 
core   0: 0x00000000800029e4 (0x00e7a023) sw      a4, 0(a5)
core   0: exception trap_store_access_fault, epc 0x00000000800029e4
core   0:           tval 0x0000000020000000
: 
core   0: >>>>  trap_entry
core   0: 0x0000000080000170 (0xef010113) addi    sp, sp, -272
core   0: 3 0x0000000080000170 (0xef010113) x2  0x0000000080022a30
```
I also defined memory parts like below:
`spike --log-commits -l -d -m0x80000000:0x8192,0x20000000:0x128 hword.riscv`
but it again go to the trap_entry. `exception trap_store_access_fault, epc 0x00000000800027a8`
Could you help me with it?
What is the reason that I am getting this error? How to solve it?
note: when I deleted the lines that writing to the memory it again goes trap_entry.

scottj97, 2022-05-20:
Spike has limited memory ranges and takes an access fault when accessing a physical address outside of these memories.

You solved the first problem the correct way. (By adding memory where you wanted it at 0x20000000.) Although 0x8192 and 0x128 are very weird sizes.

Your second problem is likely similar, but without seeing the trace around pc 800027a8 I can't tell what address it's trying to access.

omerguzelelectronicguy, 2022-05-23:
@scottj97 Thank you for reply.
As I understand, it is accessing `0x0000000080022bd0 + 128`. The instructions around it are below.
```
  core   0: >>>>  _init
  core   0: 0x000000008000279c (0xf5010113) addi    sp, sp, -176
  core   0: 3 0x000000008000279c (0xf5010113) x2  0x0000000080022b50
  : 
  core   0: 0x00000000800027a0 (0x00000797) auipc   a5, 0x0
  core   0: 3 0x00000000800027a0 (0x00000797) x15 0x00000000800027a0
  : 
  core   0: 0x00000000800027a4 (0x44878793) addi    a5, a5, 1096
  core   0: 3 0x00000000800027a4 (0x44878793) x15 0x0000000080002be8
  : 
  core   0: 0x00000000800027a8 (0x09413023) sd      s4, 128(sp)
  core   0: exception trap_store_access_fault, epc 0x00000000800027a8
  core   0:           tval 0x0000000080022bd0
  : 
  core   0: >>>>  trap_entry
  core   0: 0x0000000080000170 (0xef010113) addi    sp, sp, -272
  core   0: 3 0x0000000080000170 (0xef010113) x2  0x0000000080022a40
  : 
  core   0: 0x0000000080000174 (0x00113423) sd      ra, 8(sp)
  core   0: exception trap_store_access_fault, epc 0x0000000080000174
  core   0:           tval 0x0000000080022a48
``` These are from the simulation that I specify the memory by giving this command: `spike --log-commits -l -d -m0x80000000:0x8192,0x20000000:0x128 hword.riscv`

it may not allow to access that address but when I don't specify the memory location it behaves differently. I mean when I give that command: `spike --log-commits -l -d  hword.riscv`, it doesn't give any error on this. The sim results around it are below:
```
  core   0: >>>>  _init
  core   0: 0x000000008000279c (0xf5010113) addi    sp, sp, -176
  core   0: 3 0x000000008000279c (0xf5010113) x2  0x0000000080022b50
  : 
  core   0: 0x00000000800027a0 (0x00000797) auipc   a5, 0x0
  core   0: 3 0x00000000800027a0 (0x00000797) x15 0x00000000800027a0
  : 
  core   0: 0x00000000800027a4 (0x44878793) addi    a5, a5, 1096
  core   0: 3 0x00000000800027a4 (0x44878793) x15 0x0000000080002be8
  : 
  core   0: 0x00000000800027a8 (0x09413023) sd      s4, 128(sp)
  core   0: 3 0x00000000800027a8 (0x09413023) mem 0x0000000080022bd0 0x0000000000000000
  : 
  core   0: 0x00000000800027ac (0x00000a17) auipc   s4, 0x0
  core   0: 3 0x00000000800027ac (0x00000a17) x20 0x00000000800027ac
  : 
  core   0: 0x00000000800027b0 (0x43ca0a13) addi    s4, s4, 1084
  core   0: 3 0x00000000800027b0 (0x43ca0a13) x20 0x0000000080002be8
```

omerguzelelectronicguy, 2022-05-23:
When I increase the memory to include the unauthorized address too, it works. However, I don't know if it is a proper way to do that. On the other hand, Should I include them one by one if it gives many memory access errors?

scottj97, 2022-05-23:
> These are from the simulation that I specify the memory by giving this command: `spike --log-commits -l -d -m0x80000000:0x8192,0x20000000:0x128 hword.riscv`

I believe Spike will round up memory sizes to the nearest page (4 KiB) boundary. So you will get a memory at 0x80000000 of size 0x9000.

Did you mean 0x2000 a.k.a. (decimal) 8192?

omerguzelelectronicguy, 2022-05-24:
> Did you mean 0x2000 a.k.a. (decimal) 8192?

No, it is any numbers that I tried to include at least the whole program. I noticed that Spike round the amount of size but I forget to change `0x8192`. Excuse me if it causes misunderstanding.

> The default memory in Spike is -m0x80000000:0x80000000 if I'm reading it correctly. (2GB)

That was the answer for me. I understand why it behaves differently when I didn't specify the location. I thought very differently.
Thank you so much @scottj97 . 
I can close the issue now.

closed 2022-05-24 by omerguzelelectronicguy