riscv-software-src/riscv-isa-sim #1010: Fail run on Questasim libfesvr.so cannot open sahred object

FranDuqueAyachi, 2022-05-24:
Hi!
I'm trying to simulate the hello.elf on questasim with the command:
make sim elf-bin=$RISCV/riscv64-unknown-elf/bin/pk target-options=hello.elf batch-mode=1
My Questasim version 2019.2 Linux_x86_64
It seems to run and start loading all modules "work."
then it generate the UVM-1.1d enviroment generating the file trace_hart_o.log with all the instructions and after 40000000 ns it ends but with 2 errors wich are:
Loading /home/users/PFC/divsicad/fduque/tfg/riscv/lib/libfesvr.so

** Error (suppressible): (vsim-3197) Load of "/home/users/PFC/divsicad/fduque/tfg/riscv/lib/libfesvr.so" failed: /home/users/PFC/divsicad/fduque/tfg/riscv/lib/libfesvr.so: cannot open shared object file: No such file or directory.
when installed libfesvr by ./ci/install-libfesvr it didnt install any file .so
How can I fix this error, thank you so much!

aswaterman, 2022-05-24:
This isn’t a use case we support. I recommend you ask for assistance from whoever suggested you use libfesvr and questasim together.

scottj97, 2022-05-24:
It sounds like your problem is not with Spike (this repo) but with some other project that tries to integrate Spike with an RTL simulation in Questa. There is no `make sim` target or `install-libfesvr` script here.

closed 2022-05-24 by scottj97

FranDuqueAyachi, 2022-05-24:
> It sounds like your problem is not with Spike (this repo) but with some other project that tries to integrate Spike with an RTL simulation in Questa. There is no `make sim` target or `install-libfesvr` script here.

Yes, its true, 
But is the file libfesvr.so still exist ?

scottj97, 2022-05-24:
:shrug:

Whatever you are running that's supposed to build libfesvr.so is not a part of this project, so how are we Spike maintainers to know?

FranDuqueAyachi, 2022-05-24:
> 🤷
> 
> Whatever you are running that's supposed to build libfesvr.so is not a part of this project, so how are we Spike maintainers to know?

I dont know, im just asking because fesvr has been absorbed by riscv-isa-sim and many files has been changed and maybe I coud get answers about how can I install or find or build the libfesvr.so file.
Im also very new on all this