riscv-software-src/riscv-isa-sim #1011: [Bug Report] Memory access range of SPIKE

chenc6, 2022-05-24:
Hi,

One of our test cases tries to access memory that is outside the accessible memory space (e.g., 0x12033000). However, the spike does not throw an exception for this behavior. 

Based on riscv/platform.h, below are the valid memory addresses:
[image]

Below is the output from the spike simulation:
[image]

The instructions at 0x800025c0 and 0x800025c0 changes the value of sp to 0x12033000, and instructions at 0x800000e4 and 0x800000f8 try to access data near that memory address. However, no exceptions are throwed.

test case and log files are attached.
[testcase.txt](https://github.com/riscv-software-src/riscv-isa-sim/files/8766042/testcase.txt)
[spike.log](https://github.com/riscv-software-src/riscv-isa-sim/files/8766043/spike.log)

Thank you.

aswaterman, 2022-05-25:
Unfortunately, this bug report isn't actionable without clear instructions on how to reproduce the bug.  You could try sharing with us the following:
- Which commit of this repo you used
- Your `configure` options
- Your `spike` command line
- The ELF program you ran

scottj97, 2022-11-08:
Closing since no following from OP

closed 2022-11-08 by scottj97