riscv-software-src/riscv-isa-sim #1016: illegal instruction when reading vlenb using csrr

mahnaz-namazi, 2022-05-29:
Hi,
I have installed riscv-gnu-toolchain with riscv-gcc set at branch "riscv-gcc-rvv-next" to compile codes using vector extension v1.0.
I installed spike and pk from master branch. after compiling my code using "riscv32-unknown-elf-gcc", I see some "csrr a3,vlenb" through the .dis file at which spike throws exception "trap illegal instruction". I checked riscv/encoding to see if "vlenb" is defined and it is correct. it seems that it does not know about vlenb and I don't know if the problem is with spike or pk or even riscv-gnu-toolchain. I would appreciate any help.

p.s : I am using ubuntu 21.10 in VM on M1 MacBook Pro.
my gcc version : 11.2.0

gnu-toolchain : ./configure --prefix==/home/utm/opt/riscvv --with-arch=rv32imcv --with-abi=ilp32

spike build :  ../configure --prefix=/home/utm/opt/riscvv --with-varch=vlen:128,elen=32 --with-isa=rv32imcv --with-target=riscv32-unknown-elf

pk build  : ../configure --prefix=/home/utm/opt/riscvv --host=riscv32-unknown-elf --with-arch=rv32imc CC=riscv32-unknown-elf-gcc --with-abi=ilp32

[image]

aswaterman, 2022-05-29:
This probably isn't a spike bug.  `vlenb` is only accessible if `mstatus.VS` != 0, and no doubt the software isn't arranging for that to happen.

Rather than just raising the issue again on the riscv-pk repo, please debug the issue and propose a patch.  The people who maintain riscv-pk don't have time to do the debugging for you, I'm afraid.

mahnaz-namazi, 2022-05-30:
hi, 
I checked `mstatus` which was `0x80006622`  ,thus it seems that `mstatus.VS != 0` and vector instructions should be valid( I don't know if anything else should be set to enable vector extension)

[image]

As my application runs in U-mode, it can not access CSRs which can only be accessed in M-mode and that's why this exception arises . So the compiled code which uses vector extension will not work on spike/pk. am I correct?

aswaterman, 2022-05-31:
I don't know, but you might be using an older version of some piece of software.  I tried this same sort of thing on the latest Spike and latest pk, and it worked just fine:

```
$ cat test.c
#include <stdio.h>
int main()
{
  unsigned long vlenb;
  asm volatile ("csrr %0, vlenb" : "=r" (vlenb));
  printf("%lu\n", vlenb);
  return 0;
}
$ riscv64-unknown-elf-gcc -O2 test.c
$ spike --isa=rv64gcv pk a.out
bbl loader
16
$ spike --varch=vlen:128,elen:32 --isa=rv64gcv pk a.out
bbl loader
16
```

mahnaz-namazi, 2022-06-02:
Thanks Andrew for your answer. 

the same test you did had the following results on my machine, the only difference is that you are using 64-bit and I am using 32-bit ISA. I can not figure out what is making this error.

[image]

aswaterman, 2022-06-02:
I recommend that you try to debug the problem and submit a pull request with a bugfix.

AliAqdas-repo, 2023-01-18:
I am encountering the same issue with RV32IMV. Have you found any fix for this?

aswaterman, 2023-01-18:
It works fine for me on rv32gcv.  In any case, this is not a spike issue.

closed 2023-01-18 by aswaterman

mahnaz-namazi, 2023-01-19:
@AliAqdas-repo @aswaterman 

At the end, I had to install riscv-gnu-toolchain in **multilib** mode which supports both 32 and 64 bits. In this way I am able to compile my c codes for RV32GCV architecture and get correct results.
these are the commands I used for installing the tools:

```
git clone --recurse-submodules --remote-submodules https://github.com/riscv-collab/riscv-gnu-toolchain.git
cd riscv-gnu-toolchain/riscv-gcc
git branch -a 
git checkout riscv-gcc-rvv-next
cd .. && mkdir build && cd build
../configure --prefix=$RISCV --with-multilib-generator="rv32imc-ilp32--;rv32gcv-ilp32--"
make
```

and for installing spike:

`cd.. && make build-sim sim=spike`

aswaterman, 2023-01-19:
Thanks for sharing your solution!

AliAqdas-repo, 2023-01-19:
Thank you for sharing the workflow!