riscv-software-src/riscv-isa-sim #1019: Issue in pmpaddr0

UmerShahidengr, 2022-06-02:
Hello,
I wrote a test in riscof environment to verify the lock-bit functionality of PMP. If PMP entry i is locked, writes to pmpicfg and pmpaddri should have been ignored according to RISC-V privilege manual.
In the test, I locked the lock-bit of pmpcfg0 register in m-mode, and try to write pmpcfg0, pmpaddr0, pmpaddr1.. All these write accesses should have been blocked but pmpaddr0 is getting updated in DUT-Spike model.
The test file, DUT-Spike-Signature file, and REF-Sail-Signature file are attached.

[PMP-CFG-reg_locked.txt](https://github.com/riscv-software-src/riscv-isa-sim/files/8822313/PMP-CFG-reg_locked.txt)

[DUT-Spike-signature.txt](https://github.com/riscv-software-src/riscv-isa-sim/files/8822309/DUT-Spike-signature.txt)

[Reference-sail_c_simulator-signature.txt](https://github.com/riscv-software-src/riscv-isa-sim/files/8822305/Reference-sail_c_simulator-signature.txt)

aswaterman, 2022-06-02:
Please provide an easy way for us to replicate this bug; and, if you're convinced it is, in fact, a bug, we request that you supply a pull request to fix it.

UmerShahidengr, 2022-06-02:
In order to replicate this error, you may run this simple code,

//////////////////// TEST START HERE  /////////////////////////////////////////////

    	li	a5,-1                   // PUT 0xFFFFFFFF

	
	csrw pmpcfg0, a5		// WRITE pmpcfg0 with ALL 1s, Locked the lock-bit [7,15,23,31]
	nop
	
	csrr a5, pmpcfg0		// READ pmpcfg0		

	// Verify that LOCKED bits are HIGH by displaying it in debugger
	

	// TRY TO WRITE ADDRESS REGISTER. THIS WRITE SHOULD HAVE BEEN IGNORED 
	csrw pmpaddr0, a5		// WRITE pmpaddr0 with some other values
	nop
	
	csrr a5, pmpaddr0		// READ pmpaddr0		

	// The updated write should have been ignored and should have given the previously saved value!!!

scottj97, 2022-06-02:
I suspect what you are seeing is not broken locking, but an unexpected initial value in `pmpaddr0`. See [this code](https://github.com/riscv-software-src/riscv-isa-sim/blob/901fedb22cbf5de22fab60a6c5d7236e350ecf64/riscv/processor.cc#L503-L508) that initializes `pmpaddr0` with a different value than the others.

aswaterman, 2022-06-02:
(which, by the way, is legal, because of the subordinate clause in https://github.com/riscv/riscv-isa-manual/blob/3e7b389772b5d909b3dfa2db57e60f73329e5721/src/machine.tex#L2883-L2884)

UmerShahidengr, 2022-06-04:
Thanks @scottj97 . It was indeed an issue of unexpected initial values in different models. Thanks @aswaterman for your swift response.

closed 2022-06-04 by UmerShahidengr