riscv-software-src/riscv-isa-sim #1020: Compilation fails against updated riscv-opcodes

kwalker27, 2022-06-02:
Due to the changes from  https://github.com/riscv/riscv-opcodes/pull/106 Spike no longer compiles against the header generated by `riscv-opcodes` since pseudo-op mask/match are no longer generated into encoding.out.h:

> The new encoding.h generated from the PR will no longer include pseudo-ops as was
done earlier. This is because there exist pseudo-ops which have the same mnemonics as the regular
instruction but differ is just few bits. Eg. slli in RV32 and RV64 have the same mnemonic is both,
however in RV32 slli is treated as a pseudo_op of RV64::slli with bit 25 set to 0. Thus it is
impossible to include both variants of the instruction. Also spike doesn't make use the
pseudo-ops.

Reverting commit `ff645fb4` for #1000 is sufficient to fix the build failure, but I'm not suggesting that's the right fix. Maybe at least _some_ of the pseudo-ops should continue to end up in the header.

kwalker27, 2022-06-02:
Perhaps the "issue" belongs in `riscv-opcodes` but I decided to here, since it seems to have been an intentional change over there.

closed 2022-06-02 by aswaterman

kwalker27, 2022-06-02:
Interesting. It seems like this just allows for additional divergence until some time when Spike _needs_ an updated `encoding.h` for some reason - at which point any issues would have to be resolved one way or another?

scottj97, 2022-06-02:
I tend to agree with @kwalker27 here; we should not allow known divergences to fester because it leads to messes like #848 when we finally do have a need to update from `riscv-opcodes`.

kwalker27, 2022-06-02:
FWIW, removing the $pseudo_op tagging of the prefetch instructions and pause is also sufficient to resolve this. Again, that may not be an appropriate change but it doesn't appear to break the `riscv-opcodes` build to do this.

neelgala, 2022-06-09:
Just an FYI 

- removing $pseudo_op from those instructions is catastrophic - it breaks the definition of pseudo_ops. 
- Enabling all pseudo_ops is also an issue because instructions like slli, srli and srai in RV32 are pseudo_ops of instructions in RV128  which have the same mnenomics - so the dictionary in riscv-opcodes breaks (because multiple entries with the same key cannot exist in a python dict).
- we can rename pseudo_ops by postfixing the extension to them (e.g. MATCH_SLLI_RV32_I). But this too breaks because we have pseudo ops like fsri in multiple extensions which will cause redundant entries like (MATCH_FSRI_RV32_ZBT and MATCH_FSRI_RV32_ZBPBO).

The only way forward I see is to explicitly provide a list of pseudo_ops to the create_inst_dict function for encoding.out.h generation. I have this tested on spike and will be raising a PR for the same on riscv-opcodes.

Thanks for catching this.

aswaterman, 2022-06-09:
Resolved by https://github.com/riscv/riscv-opcodes/pull/126