riscv-software-src/riscv-isa-sim #1021: Implementing custom csr

xydas97, 2022-06-03:
Hello, I am trying to implement a custom csr by modifying the following files:
encoding.h
#define CSR_DEMO 0x404
DECLARE_CSR(democsr, CSR_DEMO)

processor.h   -   struct state_t
csr_t_p democsr;

processor.cc   -   void state_t::reset(processor_t* const proc, reg_t max_isa)
csrmap[CSR_DEMO] = democsr = std::make_shared<basic_csr_t>(proc, CSR_DEMO, 0);

But I keep getting error for unknown csr democsr.
Do I miss something or there is another file I should modify?

Thank you

xydas97, 2022-06-04:
I tried to match the name of the CSR and #define but didnt work.
I am trying to compile a simple assembly test file and I am getting Error: unknown CSR `democsr' on line 54
53: li t0, 15
54: csrs democsr, t0

scottj97, 2022-06-04:
> I tried to match the name of the CSR and #define but didnt work.
> I am trying to compile a simple assembly test file and I am getting Error: unknown CSR `democsr' on line 54
> 53: li t0, 15
> 54: csrs democsr, t0

That error is coming from the assembler, not from Spike. The assembler doesn't understand your new CSR.

There's two ways to fix that: (1) compile a custom GNU binutils with your custom CSR added; or (2) use the CSR number instead of name in the assembly source file.

For (2) you can do something like this in your `foo.S` (make sure the file extension is capital S so it will go through the preprocessor):
```
#define democsr 0x404

csrs democsr,t0
```

It might also be possible in a lowercase-s `foo.s` to use `.equ` like so:
```
.equ democsr, 0x404

csrs democsr,t0
```

xydas97, 2022-06-05:
I tried the first way and it worked. I added my custom CSR in riscv-opc.h of riscv-binutils.
Thank you.

closed 2022-06-05 by xydas97