riscv-software-src/riscv-isa-sim #1022: 48bit illegal instruction behavior

neelgala, 2022-06-03:
I have a test which checks for illegal instruction behavior which is running on rv32imc config of spike (`spike --log ref.dump --log-commits --priv=mu --isa=rv32imc ref.elf`). The following is the test-case of interest:

```
core   0: 0x80000048 (0x430000003f) unknown
core   0: exception trap_illegal_instruction, epc 0x80000048
core   0:           tval 0x00000000
```

I wanted to understand why tval is `0x00000000` instead of `0x0000003f`?

My understanding so far is that the instruction size and csr sizes are both uint64_t throughout. And for an rv32-config system it should be reading the lower 32-bit of the csr . What am I missing ?

neelgala, 2022-06-03:
going back to spec I see the following which indicates the mtval should be `0x000003f`

> If mtval is written with a nonzero value when an illegal-instruction exception occurs, then mtval will contain the shortest of:
> • the actual faulting instruction
> • the first ILEN bits of the faulting instruction
> • the first MXLEN bits of the faulting instruction

But then there is an immediate commentary as below which could indicate setting the value of mtval to 0: 

> A value of zero in mtval signifies either that the feature is not supported, or an illegal zero instruction was fetched.

Which of the above scenarios should be applied when ??

aswaterman, 2022-06-03:
I think it should be legal to write 0 some of the time, but it still looks like a Spike bug.

aswaterman, 2022-06-04:
@scottj97 I need some C++ help.  I tracked the problem down and observe the following makes the bug disappear.

To me, this reeks of undefined behavior somewhere.  What is it about explicitly creating the object then throwing it, vs. constructing in the throw statement, that makes a difference?  https://github.com/riscv-software-src/riscv-isa-sim/commit/32e925d6e12fbdd2d101c002c993131c23575338

aswaterman, 2022-06-04:
Perhaps the issue is that the object’s lifetime only lasts til the end of the statement? If so, I’ll need to audit all throw statements. I guess I really don’t know C++.

neelgala, 2022-06-04:
> A value of zero in mtval signifies either that the feature is not supported, or an illegal zero instruction was fetched.

Sorry for treading on this a little more - but are I find the two statements `the first ILEN bits of the faulting instruction` and `A value of zero in mtval signifies either that the feature is not supported` conflicting. Are both behaviors considered legal ? Is this an implementation-defined choice ?

If I implement only rv64gc subset (ilen for which does not cross 32-bits) and I encounter `0x00003f` instruction - should the value in mtval be just the lower 32-bits or should it be the whole 48-bits starting from that instruction?

aswaterman, 2022-06-04:
Let’s separate the Spike issue from the spec issue. There’s a bug in Spike, which we’ll fix.

The spec does seem to give ambiguous guidance, but that’s a topic for a different repository than this one. @jhauser-us told me he’d propose alternate wording in the next several weeks.

scottj97, 2022-06-04:
> @scottj97 I need some C++ help. I tracked the problem down and observe the following makes the bug disappear.
> 
> To me, this reeks of undefined behavior somewhere. What is it about explicitly creating the object then throwing it, vs. constructing in the throw statement, that makes a difference? [32e925d](https://github.com/riscv-software-src/riscv-isa-sim/commit/32e925d6e12fbdd2d101c002c993131c23575338)

I see no problem with the current code, and it conforms to [some C++ best practices](https://isocpp.org/wiki/faq/exceptions#what-to-throw).

Compiler bug?

aswaterman, 2022-06-07:
Figured it out.  In `insn_t::bits()`, we are left-shifting by greater than the datatype width for 8-byte instructions.  Since it's UB, it manifests only some of the time, depending on the compiler's choices.

aswaterman, 2022-06-07:
Looks like this bug was introduced in ef3ea15aba0dc0a9f0fde85f1e21c8b98d76ace2.  It's tempting just to fix it directly, but it might be better just to remove the need for this sort of masking altogether.

commit dbfa8aff (2022-06-07): Don't mask instruction bits

No longer needed, since they are no longer sign-extended.

Fixes #1022 by eliminating undefined behavior (64-bit instructions
resulted in a shift amount equal to the datatype width).
commit 5ada8452 (2022-06-07): Don't mask instruction bits

No longer needed, since they are no longer sign-extended.

Fixes #1022 by eliminating undefined behavior (64-bit instructions
resulted in a shift amount equal to the datatype width).
commit 7d943d74 (2022-06-07): Don't mask instruction bits

No longer needed, since they are no longer sign-extended.

Fixes #1022 by eliminating undefined behavior (64-bit instructions
resulted in a shift amount equal to the datatype width).
commit ac466a21 (2022-06-07): Merge pull request #1027 from riscv-software-src/fix-1022

Fix #1022 by removing need to mask insn_t bits
closed 2022-06-07 by aswaterman