riscv-software-src/riscv-isa-sim #1028: why cannot open pk?

yuyang-ok, 2022-06-11:
~~~
yuyang@yuyang-vmwarevirtualplatform:~/projects/riscv-gcc-test$ ../riscv-isa-sim/build/spike  pk ./a.out 
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not open pk (did you misspell it? If VCS, did you forget +permissive/+permissive-off?)
~~~

yuyang-ok, 2022-06-11:
[hello.zip](https://github.com/riscv-software-src/riscv-isa-sim/files/8882940/hello.zip)

aswaterman, 2022-06-11:
Did you install pk?

yuyang-ok, 2022-06-12:
pk?? what is pk? I think pk is a command argument. no need to install.

aswaterman, 2022-06-12:
pk is a program that you’re attempting to run but that you don’t have installed. You can find it here: https://github.com/riscv-software-src/riscv-pk

Closing because there’s no Spike bug evident.

closed 2022-06-12 by aswaterman